bolderflight/invensense-imu #4: Add calibration functions

aster94, 2017-11-25:
It would be quite useful to have them
I think you should add something to calibrate while running the sketch,  a function to read the calibration data and a function to write calibration data

flybrianfly, 2017-12-13:
Calibration functions for the gyroscopes, accelerometer, and magnetometer have been added. Functions to retrieve the estimated bias and scale factor for each sensor, for each axis have also been added as well as functions to set the bias and scale factor being used. These functions should make it easy for the user to calibrate the sensor, retrieve the calibration parameters, store them (i.e. in the microcontroller EEPROM), and apply those calibration parameters during future uses.

I chose not to use the MPU-9250 built in calibration storage for two reasons:

1.  It only supports the gyroscope and accelerometer.
2.  It only supports saving and applying a bias.

The gyroscope calibration results only in needing a bias, but the gyroscopes should be calibrated every time the sensor is used. So gyroscope calibration is now part of the *begin* function. Accelerometer and magnetometer calibrations do not need to be performed as frequently, meaning storage makes sense, but require both a bias and a scale factor. It seems like using the microcontroller's EEPROM makes the most sense for storing these values.

closed 2017-12-13 by flybrianfly

aster94, 2017-12-14:
Hello Brian,
This post is just to be sure (or clarify for future users) how the calibration functions work, tell me if i wrote something wrong.
For example if i have a mpu9250 that later will be mounted in a device (robot, motorbike, quadcopter,ecc), i will do:

1. run calibrateGyro, accel and mag
2. get (and store) the bias of the two sensors using the functions: getAccelBiasX_mss(), getMagBiasX_uT(), for all axis
3. mount the device with the imu in its final position
4. in the setup use: setAccelBiasX_mss(), setMagBiasX_uT(), ecc using the values i get at point 2

but this brings me to two questions:
1) what happens if i **can't** mount the device in a flat position? running the calibrateGyro() inside the begin will brings to errors? wouldn't it be better if i use the getGyroBiasX_rads() while i get the other bias and then use setGyroBiasX_rads() instead of using the values i get from the askew sensor?
2) is there any difference in the get/set functions in using the correction with mss/factor for the accelerometer or uT/factor for the magnetomer?

flybrianfly, 2017-12-16:
In response to your questions:

1. For calibrateGyro() to estimate the gyro biases correctly, the IMU just needs to be motionless, not necessarily level. Additionally, gyro bias is highly sensitive to temperature, which is why I recommend calibrating gyros every time the sensor is used rather than storing and using a static set of gyro biases.
2. There is no difference between the functions used for get and set other than the memory locations being accessed.

Some general information:

1. Accelerometer calibration only needs to be done once.
2. Magnetometer calibration should be done in the final installation location with everything powered and running, if possible. Any changes in the magnetic or electrical environment would suggest performing a new magnetometer calibration.

I recently posted a 7 state EKF AHRS Arduino library, which requires a well calibrated IMU to work well. As part of the example included with the library is an [IMU calibration sketch](https://github.com/bolderflight/uNavAHRS/tree/master/examples/uNavAHRS-with-MPU9250/CalibrateMPU9250).

aster94, 2018-01-08:
hey Brian, have a good new year,

i am running a version of CalibrateMPU9250.ino without eempro write:

`#include "Streaming.h"
#include "MPU9250.h"
#define SS_PIN PB12
SPIClass mySPI (2);
MPU9250 Imu(mySPI, SS_PIN);
int status;
float value;

void setup() {
  // serial to display instructions
  Serial.begin(115200);
  while (!Serial) {}
  // start communication with IMU
  status = Imu.begin();
  if (status < 0) {
    Serial.println("IMU initialization unsuccessful");
    Serial.println("Check IMU wiring or try cycling power");
    Serial.print("Status: ");
    Serial.println(status);
    while (1) {}
  }
  // calibrating accelerometer
  Serial.println("Starting Accelerometer Calibration");
  Imu.calibrateAccel();
  Serial.println("Switch");
  delay(5000);
  Imu.calibrateAccel();
  Serial.println("Switch");
  delay(5000);
  Imu.calibrateAccel();
  Serial.println("Switch");
  delay(5000);
  Imu.calibrateAccel();
  Serial.println("Switch");
  delay(5000);
  Imu.calibrateAccel();
  Serial.println("Switch");
  delay(5000);
  Imu.calibrateAccel();
  Serial.println("Done");
  Serial.println("Starting Magnetometer Calibration");
  delay(5000);
  // calibrating magnetometer
  //move with a 8 shape
  Imu.calibrateMag();

  value = Imu.getAccelBiasX_mss();
  Serial << "Accel bias x " << value << newl;
  value = Imu.getAccelScaleFactorX();
  Serial << "Accel factor x " << value << newl;
  value = Imu.getAccelBiasY_mss();
  Serial << "Accel bias y " << value << newl;
  value = Imu.getAccelScaleFactorY();
  Serial << "Accel factor y " << value << newl;
  value = Imu.getAccelBiasZ_mss();
  Serial << "Accel bias z " << value << newl;
  value = Imu.getAccelScaleFactorZ();
  Serial << "Accel factor z " << value << newl;

  value = Imu.getMagBiasX_uT();
  Serial << "mag bias x " << value << newl;
  value = Imu.getMagScaleFactorX();
  Serial << "mag factor x " << value << newl;
  value = Imu.getMagBiasY_uT();
  Serial << "mag bias y " << value << newl;
  value = Imu.getMagScaleFactorY();
  Serial << "mag factor y " << value << newl;
  value = Imu.getMagBiasZ_uT();
  Serial << "mag bias z " << value << newl;
  value = Imu.getMagScaleFactorZ();
  Serial << "mag factor z " << value << newl;

  Serial.println("Done");
}

void loop() {}`

the mag calibration works ok, but from the accelerometer i get bias 0 and scale factor 1 for all the three axis.
any idea why? or maybe my sensor is perfect ahahahah

flybrianfly, 2018-01-08:
Are you getting all 6 orientations of the IMU (+x, -x, +y, -y, +z, -z) aligned with the gravity vector (i.e. pointed down)? The code checks to make sure that you got all of those before computing a scale factor and bias.

aster94, 2018-01-12:
don't know how but i was doing something wrong, is this sequence  (+x, -x, +y, -y, +z, -z)  important? or i could follow any (example:  +y, -y, +z,+x, -x, -z) ?

flybrianfly, 2018-01-13:
The sequence shouldn't be important, as long as all 6 of the orientations are aligned with the gravity vector.

iFrostizz, 2019-08-02:
Hello!
Using the sketch Basic I2C, it seems that the accelerometer calibration don't work since I get zacc = -8,71 when flat.