bolderflight/invensense-imu #5: readSensor function don't work in interrupt

ZahraHeidari, 2017-12-04:
I'm calling readSensor function in interrupts (first in setup() call enableDataReadyInterrupt()) and using I2C connection. interrupt is received and go to function but then it hangs and running code is stoped. why did this happen?

#include "MPU9250.h"

// an MPU9250 object with the MPU-9250 sensor on I2C bus 0 with address 0x68
MPU9250 IMU(Wire,0x68);
int status;

void setup() {
  // serial to display data
  Serial.begin(115200);
  while(!Serial) {}
  // start communication with IMU 
  status = IMU.begin();
  if (status < 0) {
    Serial.println("IMU initialization unsuccessful");
    Serial.println("Check IMU wiring or try cycling power");
    Serial.print("Status: ");
    Serial.println(status);
    while(1) {}
  }
  else Serial.println("initiate ok");
  // setting the accelerometer full scale range to +/-16G 
  IMU.setAccelRange(MPU9250::ACCEL_RANGE_16G);
  // setting the gyroscope full scale range to +/-500 deg/s
  IMU.setGyroRange(MPU9250::GYRO_RANGE_500DPS);
  // setting DLPF bandwidth to 20 Hz
  IMU.setDlpfBandwidth(MPU9250::DLPF_BANDWIDTH_20HZ);
  // setting SRD to 19 for a 50 Hz update rate
  IMU.setSrd(19);
  // enabling the data ready interrupt
  IMU.enableDataReadyInterrupt();
  // attaching the interrupt to microcontroller pin 1
  pinMode(2,INPUT);
  attachInterrupt(digitalPinToInterrupt(2),getIMU,RISING);
}

void loop() {
  // display the data
  Serial.println("loop");
/*  Serial.print(IMU.getAccelX_mss(),6);
  Serial.print("\t");
  Serial.print(IMU.getAccelY_mss(),6);
  Serial.print("\t");
  Serial.print(IMU.getAccelZ_mss(),6);
  Serial.print("\t");
  Serial.print(IMU.getGyroX_rads(),6);
  Serial.print("\t");
  Serial.print(IMU.getGyroY_rads(),6);
  Serial.print("\t");
  Serial.print(IMU.getGyroZ_rads(),6);
  Serial.print("\t");
  Serial.print(IMU.getMagX_uT(),6);
  Serial.print("\t");
  Serial.print(IMU.getMagY_uT(),6);
  Serial.print("\t");
  Serial.print(IMU.getMagZ_uT(),6);
  Serial.print("\t");
  Serial.println(IMU.getTemperature_C(),6);*/
}

void getIMU(){
  // read the sensor
  IMU.readSensor();
  Serial.println("interrupts");
}

flybrianfly, 2017-12-04:
This code works fine for me on a Ladybug STM32 board with the interrupt pin changed (it doesn't have a pin 2).
1. What board are you testing on?
2. If you run the Basic_I2C example, and use micros() to time the readSensor function, how long is it taking to complete?

ZahraHeidari, 2017-12-06:
I'm using Arduino Mega. it takes about 110000 micro sec to run readSensor function and print them.

flybrianfly, 2017-12-13:
I think your interrupt is running too fast in that case and unfortunately, the SRD doesn't support sample rates slow enough to run the function. You may want to consider setting up your own timer using millis() to read the sensor.

mrbubble62, 2018-01-21:
Don't do any work in the interrupt, just set a flag

```
volatile boolean flagDataReady;  // interrupt flag

// read interrupt handler
void getIMU()
{
	flagDataReady = true;
}

```
then in your main loop 
```
if (flagDataReady) { // check for interrupt
	IMU.readSensor();
	//reset interrupt flag
	flagDataReady = false;
}
```

closed 2018-07-26 by flybrianfly