bolderflight/invensense-imu #6: Can't compile for Arduino Nano nor Arduino Uno

lgenzelis, 2018-01-06:
When trying to compile for the Arduino Nano or the Arduino Uno boards, I get the error 
`Global variables use 5368 bytes (262%) of dynamic memory, leaving -3320 bytes for local variables. Maximum is 2048 bytes.`

So, is this library not compatible with these boards? In that case, it should be mentioned in the docs.

flybrianfly, 2018-01-07:
Yes, it looks like it won't work as is for the Nano or Uno due to a lack of memory. Sorry, we don't have those boards for testing.

Thanks for looking into the FIFO, we'll have to look at options there to reduce memory usage if the FIFO is not being used (i.e. dynamic allocation, a separate class, etc).

lgenzelis, 2018-01-07:
You are welcome! IMO a separate class would be the simplest option.

flybrianfly, 2018-01-17:
I split the functionality into a base MPU9250 class and a derived MPU9250FIFO class. This greatly reduces the memory footprint for the base class. I compiled an example using the base class for the Uno and Nano; both fit within the memory requirements of the boards. I would appreciate any feedback from users using these boards to confirm that the library works well with them; we only have Teensy 3.x, LC, and Ladybug microcontrollers to test with at the moment.

closed 2018-01-17 by flybrianfly

lgenzelis, 2018-01-24:
@flybrianfly tested the new version with arduino nano. Works like a charm!

BigBernie23, 2019-01-14:
I'm sorry guys.  I've downloaded the latest library and I'm still getting the same problem on my Nano.
Minimum Memory Usage: 5142 bytes (251% of a 2048 byte maximum)
The only gap, if that's the issue, is that I'm using wire instead of SPI for the FIFO declaration

MPU9250FIFO IMU(Wire, 0x68);

Any ideas?  Thanks,

flybrianfly, 2019-01-14:
The FIFO won't work on the nano, there isn't enough memory. This is why I split the library into a base MPU9250 and a derived MPU9250FIFO class - so users with nano's and other low memory Arduino boards can use the base MPU9250 class.

BigBernie23, 2019-01-15:
Thanks Brian for the promptly answer.  Bests,