bolderflight/invensense-imu #7: Can't compile the basic SPI example

Thzn, 2018-01-15:
Hi,

I'm using Arduino 1.8.5
I'm trying to compile https://github.com/bolderflight/MPU9250/blob/master/examples/Basic_SPI/Basic_SPI.ino code. but I got a compile error,

Basic_SPI:27: error: 'SPI' was not declared in this scope
 MPU9250 IMU(SPI,10);
             ^
/tmp/arduino_modified_sketch_937275/Basic_SPI.ino: In function 'void setup()':
Basic_SPI:36: error: 'class MPU9250' has no member named 'begin'
   status = IMU.begin();
                ^
/tmp/arduino_modified_sketch_937275/Basic_SPI.ino: In function 'void loop()':
Basic_SPI:48: error: 'class MPU9250' has no member named 'readSensor'
   IMU.readSensor();
       ^
Basic_SPI:50: error: 'class MPU9250' has no member named 'getAccelX_mss'
   Serial.print(IMU.getAccelX_mss(),6);
                    ^
Basic_SPI:52: error: 'class MPU9250' has no member named 'getAccelY_mss'
   Serial.print(IMU.getAccelY_mss(),6);
                    ^
Basic_SPI:54: error: 'class MPU9250' has no member named 'getAccelZ_mss'
   Serial.print(IMU.getAccelZ_mss(),6);
                    ^
Basic_SPI:56: error: 'class MPU9250' has no member named 'getGyroX_rads'
   Serial.print(IMU.getGyroX_rads(),6);
                    ^
Basic_SPI:58: error: 'class MPU9250' has no member named 'getGyroY_rads'
   Serial.print(IMU.getGyroY_rads(),6);
Multiple libraries were found for "MPU9250.h"
                    ^
 Used: /home/tk/Arduino/libraries/MPU9250-master
Basic_SPI:60: error: 'class MPU9250' has no member named 'getGyroZ_rads'
 Not used: /home/tk/Arduino/libraries/MPU9250-new
   Serial.print(IMU.getGyroZ_rads(),6);
                    ^
Basic_SPI:62: error: 'class MPU9250' has no member named 'getMagX_uT'
   Serial.print(IMU.getMagX_uT(),6);
                    ^
Basic_SPI:64: error: 'class MPU9250' has no member named 'getMagY_uT'
   Serial.print(IMU.getMagY_uT(),6);
                    ^
Basic_SPI:66: error: 'class MPU9250' has no member named 'getMagZ_uT'
   Serial.print(IMU.getMagZ_uT(),6);
                    ^
Basic_SPI:68: error: 'class MPU9250' has no member named 'getTemperature_C'
   Serial.println(IMU.getTemperature_C(),6);
                      ^
exit status 1
'SPI' was not declared in this scope

can you please help me. 

Thank You.

flybrianfly, 2018-01-15:
Which Arduino board are you compiling for?

Thzn, 2018-01-16:
It is an Arduino Mega Board.

[image]

I attached the screenshot as well.
Thank You.

flybrianfly, 2018-01-16:
I installed Arduino 1.8.5 and am able to compile for Arduino Mega with no warnings or errors. Maybe try reinstalling Arduino? You could also check whether the SPI library is present; it should be in your arduino installation folder under /hardware/arduino/avr/libraries.

closed 2018-07-26 by flybrianfly