bolderflight/invensense-imu #9: MPU9255 Data is wrong and recommend some improvements

williamesp2015, 2018-01-21:


flybrianfly, 2018-01-21:
Can you please attach or send the code you are using for the counts data?

For the SPI methods, you can use the _SPI.setMISO_, _SPI.setMOSI_, and _SPI.setSCK_ commands, like this:

`MPU9250 IMU(SPI,10);
int status;

void setup() {
  // serial to display data
  Serial.begin(115200);
  while(!Serial) {}

  SPI.setMOSI(7);
  SPI.setMISO(8);
  SPI.setSCK(14);

  // start communication with IMU 
  status = IMU.begin();
  if (status < 0) {
    Serial.println("IMU initialization unsuccessful");
    Serial.println("Check IMU wiring or try cycling power");
    Serial.print("Status: ");
    Serial.println(status);
    while(1) {}
  }
}`

williamesp2015, 2018-01-23:
Thanks. I am using ESP32 development board and it has not SPI.setMOSI.... so I mannually configured SPI pins then start SPI (SPI.begin(_SCK,_MISO, _MOSI, ss);). I have doubt about my GY-9255 MPU-9255 Sensor Module (http://www.electrodragon.com/product/9dof-mpu-9255-3d-acceleromagnetogyroscope-breakout/). Anyone had experience with this board?

Ananfruit, 2020-10-26:
Can you share the code when you calibrate your accel and gyro, please? Thanks in advance!

closed 2021-02-14 by flybrianfly