bolderflight/invensense-imu #10: TwoWire does not name a type

CoWayger, 2018-01-23:
Hello, I'm using this library on arduino 1.8.5 but master branch gives me this error:

> MPU9250.h:123:5: error: 'TwoWire' does not name a type
>      TwoWire *_i2c;
>      ^
> exit status 1

flybrianfly, 2018-01-23:
Hello, what board are you compiling for? It sounds like it either does not include an I2C library by default or is following a non-standard naming convention for the I2C library.

CoWayger, 2018-01-23:
It is STM32F103F8 board. It has SPI and I2C working (via SPI.h & I2C.h).

flybrianfly, 2018-01-23:
Which core are you using for the board? Can you point to the github repo for it? I'd like to take a look and there are several STM32 arduino cores floating around. Thanks!

CoWayger, 2018-01-23:
https://github.com/rogerclarkmelbourne/Arduino_STM32

flybrianfly, 2018-01-23:
Thanks. Can you check that you're using the newest version of that core? It was modified two months ago to be in line with the Arduino standards regarding the use of TwoWire:
http://stm32duino.com/viewtopic.php?f=3&t=2932

CoWayger, 2018-01-23:
I changed TwoWire to HardWire and now it works...

CoWayger, 2018-01-23:
I reverted my changes and updated STM32... now it works. Thanks! :)

closed 2018-01-23 by flybrianfly