bolderflight/invensense-imu #11: Magnetometer readings freeze when SRD is above 34

lgenzelis, 2018-01-24:
Starting with the example script Interrupt_SPI, if I change the SRD value to something up to 34, everything goes smoothly. However, setting it to 35 or higher causes the magnetometer readings to freeze (i.e., the rest of the readings change, but not the magnetometer ones).

lgenzelis, 2018-01-25:
The method setSrd would fail with status -2 or -3 every time, meaning that the problem was indeed related to the configuration messages sent to the magnetometer. The problem was solved by writing these messages with an SRD of 19 and setting it to the desired value at the end of the method (pull request #12 ).

flybrianfly, 2018-01-26:
That's interesting. I can re-create the issue and the fix on I2C and SPI. I'll be incorporating those changes momentarily. Thanks!

closed 2018-01-26 by flybrianfly