bolderflight/invensense-imu #17: Motion Apps

flybrianfly, 2018-02-10:
Add AHRS and / or DMP.

bingcheng1998, 2019-05-19:
Hey! 
Thanks for your fantastic work! It helps me a lot.
I tried to add DMP to the library myself, but I failed.
Your code works well!

bingcheng1998, 2019-05-19:
[BUG] e... It looks that something went wrong...
```c++
  Serial.print(hx,6);
  Serial.print("\t");
  Serial.print(hy,6);
  Serial.print("\t");
  Serial.print(hz,6);
  Serial.print("\t");
```
h output doesn't change at all.
[image]

jedjoud10, 2020-07-26:
Would it even be possible to add DMP support? I'm not even sure that the MPU 9050 has a DMP because I see no documentation about it online at all

draeron, 2020-08-07:
not sure about the MPU9050 but the MPU9150 does, there ain't much documentation on it though.

https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU9150

flybrianfly, 2021-02-14:
The MPU-9250 does have DMP, but we're not planning to add it at this time.

closed 2021-02-14 by flybrianfly