bolderflight/invensense-imu #18: Measurements like a toothsaw

peguerosdc, 2018-02-13:
Hi!

I am trying to read a MPU9250 following the same configurations as the Advanced_I2C example (ACCEL_RANGE_8G, GYRO_RANGE_500DPS, DLPF_BANDWIDTH_20HZ, SRC=19) but I am having weird measurements. Apparently, only one of every two measurements is correct:

Accel( -0.002394, 0.603370, 0.002394)
Accel( -0.035915, 0.615341, -9.907711)
Accel( -0.002394, 2.449393, 0.002394)
Accel( -0.047886, 0.644073, -9.843065)
Accel( -0.002394, 0.687171, 0.002394)
Accel( -0.038309, 0.610553, -20.837799)

This happens the same with the magnetic field but not with the gyroscope and after a couple of minutes, my MPU stops responding.

Any ideas on why this could be happening?

**More context:** when I run the "Advanced_I2C" example as is, everything works correctly. My implementation is in code where I am reading a pressure sensor as well and sending everything (acceleration, gyroscope, magnetometer, pressure) via a LoRa transmitter.

jamesterjim, 2018-02-24:
I had the same issue. I think I just fixed it by changing the SPI speeds in MPU9250.h on line 132, 133 from 

`    const uint32_t SPI_LS_CLOCK = 1000000;  // 1 MHz
    const uint32_t SPI_HS_CLOCK = 15000000; // 15 MHz`

to 

`    const uint32_t SPI_LS_CLOCK = 1000000;  // 1 MHz
    const uint32_t SPI_HS_CLOCK = 15000000; // 15 MHz`

Ananfruit, 2020-10-26:
Can you share the code when you calibrate your accel and gyro, please? Thanks in advance!

peguerosdc, 2020-11-02:
@Ananfruit I don't have this code anymore. Sorry.

closed 2020-11-02 by peguerosdc