bolderflight/invensense-imu #19: readSensor() stops giving back values after a couple minutes

balzss, 2018-02-27:
Hi, I'm using the library with the ESP32 and it works fine for a few minutes but then it only returns -1 and I have to restart the program. I tried the basic and the advanced i2c example as well. Do you have any suggestion what could be the problem? Or if this is unavoidable is there a way to restart the imu faster? Because right now the begin function does calibration and everything which takes a couple seconds and I want it to go back reading data as fast as possible. Thanks!

balzss, 2018-02-28:
Ok, I did some debugging and rewrote the readRegisters function like so:
```c++
int MPU9250::readRegisters(uint8_t subAddress, uint8_t count, uint8_t* dest){
  if( _useSPI ){
    // SPI code left out for conciseness
  } else {
    _i2c->beginTransmission(_address); // open the device
    _i2c->write(subAddress); // specify the starting register address
    int resultEndTransmission = _i2c->endTransmission(false);
    if(resultEndTransmission > 0) {
        return -200 - resultEndTransmission;
    }
    _numBytes = _i2c->requestFrom(_address, count); // specify the number of bytes to receive
    if (_numBytes == count) {
      for(uint8_t i = 0; i < count; i++){ 
        dest[i] = _i2c->read();
      }
      return 1;
    } else {
      return -100 - _numBytes;
    }
  }
}
```

and the readSensor function like so:
```c++
int MPU9250::readSensor() {
  _useSPIHS = true; // use the high speed SPI for data readout
  // grab the data from the MPU9250
  int resultReadReg = readRegisters(ACCEL_OUT,21, _buffer);
  if (resultReadReg < 0) {
      return resultReadReg;
  }
 // rest of the function left out for conciseness
}
```

After the modifications I monitored the readSensor's return value and I got a couple -100 but it always continued flawlessly. However after a couple of minutes I started getting -205 return values which according to the arduino docs shouldn't even be possible because the endTransmission function's return values should range from 0 to 4 (and not 5).

A tried to just disconnect the sensor while running and I got the same error code but after reconnecting it the readings didn't continue.

I can imagine that some of my freezings happen because the cables disconnect for a very short time but there are times when everything is completely still and the freezing still occurs.

I also wanted to connect is via SPI but I couldn't get it working at all.

flybrianfly, 2018-02-28:
Is this the core that you're using? Are you on the newest version or something older?
https://github.com/espressif/arduino-esp32

I'll need to dig into it in detail to try to figure out what's going on. Seems like maybe something with the endTransmission(false) isn't working 100% reliably. Unfortunately, my schedule is a bit tied up until after March 9th to dig around in detail.

balzss, 2018-03-01:
Yes I'm using the latest arduino-esp32 core. Reading through [this](https://github.com/espressif/arduino-esp32/issues/811) thread it seems like that the problem is in the core itself and we can't do anything with your library to avoid it so I think my (and anyone with this problem) only options are:
- using SPI instead of I2C
- using ESP8266 instead of ESP32

I tried my luck again with the SPI version and I was able the initialize the sensor (you have to use the VSPI labeled pins instead of the SPI ones) but the reading was all over the place, like I was generating just random numbers. Do you have any idea why I didn't got usable readings? I'll try to debug it later today. And another question: if the sensor is working correctly through an SPI connection and I temporarily disconnect one of the cables for a short period, will it work flawlessly after that? Or will I have to initialize the sensor again? Thanks for the help!

closed 2021-02-14 by flybrianfly