bolderflight/invensense-imu #20: Does MPU9255 FIFO can enable Interrupt?

williamesp2015, 2018-03-08:
MPU9255 has 512 bytes FIFO and always old data will be replaced with new data. I could not find a solution to trigger Interrupt once FIFO is full.
If there is not way to use Interrupt, how to check FIFO size with SPI and Arduino. 
Thanks

flybrianfly, 2018-03-08:
If you see the getFifo functions, I return the FIFO size along with the data. Do you need an interrupt or can you set a delay like the FIFO_SPI example?

williamesp2015, 2018-03-08:
Hi, _fifoSize need to check constantly and delay also cause a problem in Arduino loop() only trigger Interrupt (Arduino) can save CPU and doing something else. If there is no way then should find the duration of FIFO and read it on time.

williamesp2015, 2018-03-18:
I have solved MPU9255 FIFO Interrupt enable with this function:
void MPU9255::FIFO2(int Ascale,int Gscale,bool ACCFIFO,bool GYRFIFO,bool MAGFIFO,bool TMPFIFO,bool Fifo2INTerrupt,uint8_t SRD){
    uint16_t ii, packet_count, fifo_count;
        // reset device
    WriteReg(MPUREG_PWR_MGMT_1, 0x80); // Write a one to bit 7 reset bit; toggle reset device
    delay(100);

    // get stable time source; Auto select clock source to be PLL gyroscope reference if ready
    // else use the internal oscillator, bits 2:0 = 001
    WriteReg(MPUREG_PWR_MGMT_1, 0x01);
    WriteReg(MPUREG_PWR_MGMT_2, 0x00);
    delay(200);

    // Configure device for bias calculation
    WriteReg(MPUREG_INT_ENABLE, 0x00);   // Disable all interrupts
    WriteReg(MPUREG_FIFO_EN, 0x00);      // Disable FIFO
    WriteReg(MPUREG_PWR_MGMT_1, 0x00);   // Turn on internal clock source
    WriteReg(MPUREG_I2C_MST_CTRL, 0x00); // Disable I2C master
    WriteReg(MPUREG_USER_CTRL, 0x00);    // Disable FIFO and I2C master modes
    WriteReg(MPUREG_USER_CTRL, 0x0C);    // Reset FIFO and DMP
    delay(15);

    // Configure MPU6050 gyro and accelerometer for bias calculation
    WriteReg(MPUREG_CONFIG, 0x01);      // Set low-pass filter to 188 Hz
        // Set sample rate = gyroscope output rate/(1 + SMPLRT_DIV)
        /* setting the sample rate divider */
    //  writeRegister(MPUREG_SMPLRT_DIV,SRD); // setting the sample rate divider
      // Set sample rate = gyroscope output rate/(1 + SMPLRT_DIV)
      WriteReg(MPUREG_SMPLRT_DIV, 0);  // 0 Set sample rate to 1 kHz
      //WriteReg(MPUREG_SMPLRT_DIV, SRD);  // 0 Set sample rate to 1 kHz
    //  WriteReg(MPUREG_SMPLRT_DIV, 0x04);  // Set sample rate to 200 Hz
    ///WriteReg(MPUREG_SMPLRT_DIV, 0x00);  // Set sample rate to 1 kHz
  //  WriteReg(MPUREG_SMPLRT_DIV, 0x04);  // Set sample rate to 200 Hz
    //WriteReg(MPUREG_GYRO_CONFIG, 0x00);  // Set gyro full-scale to 250 degrees per second, maximum sensitivity
    //WriteReg(MPUREG_ACCEL_CONFIG, 0x00); // Set accelerometer full-scale to 2 g, maximum sensitivity

     gyrosensitivity  = set_gyro_scale(Gscale);   // = 131 LSB/degrees/sec
     accelsensitivity = set_acc_scale(Ascale);//16384;  // = 16384 LSB/g

    // Configure FIFO to capture accelerometer and gyro data
    WriteReg(MPUREG_USER_CTRL, 0x40);   // Enable FIFO
    //WriteReg(MPUREG_FIFO_EN, 0x78);     // Enable gyro and accelerometer sensors for FIFO  (max size 512 bytes in MPU-9150)
    //if(writeRegister(FIFO_EN,(accel*FIFO_ACCEL)|(gyro*FIFO_GYRO)|(mag*FIFO_MAG)|(temp*FIFO_TEMP)) < 0){
    //  return -2;
    //}
    uint8_t FIFOCommands=0;
    if(ACCFIFO)
    {
      FIFOCommands=FIFO_ACCEL;
    }
    if(GYRFIFO)
    {
      FIFOCommands=FIFOCommands|GYRFIFO;
    }
    if(MAGFIFO)
    {
      FIFOCommands=FIFOCommands|FIFO_MAG;
    }
    if(TMPFIFO)
    {
      FIFOCommands=FIFOCommands|FIFO_TEMP;
    }
    //const uint8_t FIFO_TEMP = 0x80; 10000000
    //const uint8_t FIFO_GYRO = 0x70; 1110000
    //const uint8_t FIFO_ACCEL = 0x08;00001000
    //const uint8_t FIFO_MAG = 0x01;  00000001
    //writeRegister(FIFO_EN,FIFO_ACCEL|FIFO_GYRO|FIFO_MAG|FIFO_TEMP) ;
    //writeRegister(FIFO_EN,FIFO_ACCEL|FIFO_GYRO|FIFO_MAG) ;
    writeRegister(FIFO_EN,FIFOCommands);
    //writeRegister(FIFO_EN,FIFO_ACCEL|FIFO_GYRO) ;
    _enFifoAccel = ACCFIFO;
   _enFifoGyro = GYRFIFO;
   _enFifoMag = MAGFIFO;
   _enFifoTemp =  TMPFIFO;
   //_fifoFrameSize = ACCFIFO*6 + GYRFIFO*6 + MAGFIFO*7 + TMPFIFO*2;
   _fifoFrameSize = ACCFIFO*6 + GYRFIFO*6 + MAGFIFO*6 + TMPFIFO*2;
  //  _fifoFrameSize =12 ;//accel*6 + gyro*6 + mag*7 + temp*2;
//512/19=26.9-->26*19=494 26msec
//512/12=42.66 ->>42*12=504 42 msec
  //  delay(42); // accumulate 40 samples in 40 milliseconds = 480 bytes
  if(Fifo2INTerrupt)
  {
        WriteReg(MPUREG_INT_ENABLE, 16);   //Bit4 =1 to enable interrupts
  }
  }

flybrianfly, 2018-03-18:
Which Interrupt settings are you using on the MPU-9250? The only related one I saw was for buffer overflow, which I was going to test, but I would assume means that you're already missing data. Register 56 in the MPU-9250 register map: https://github.com/bolderflight/MPU9250/blob/master/docs/MPU-9250-Register-Map.pdf

williamesp2015, 2018-03-30:
Hi flybrianfly. I am using FIFO Interrupt overflow Interrupt to read FIFO periodically according to the SRD and _fifoFrameSize.

Asmaa457, 2020-03-02:
i need to full the fifo then read it what is the interrupt  i will use or how can i solve this

closed 2021-02-14 by flybrianfly