bolderflight/invensense-imu #23: Error at initialisation

DomFo, 2018-03-26:
Hi and thank you for this library,

for some  reason I encounter an error when calling "IMU.begin()". The status returns -18, so writing the control register 1 of the magnetometer to continuous measurement mode does not work.

Also, if I uncomment this section from the cpp file, the next error occurs at the gyro-calibration (IMU.begin() returns -20). 

Do you have any idea what that could be based on?

Thank you for any help. 

Dom

flybrianfly, 2018-04-02:
Hi Dom,

What breakout board and Arduino are you trying to use?

Thanks,
Brian

DomFo, 2018-04-03:
Hi Brian,

I'm using the Teensy3.6 with the sparkfun breakout board. 

Meanwhile I'm using the SPI interface facing the same error as soon as I alter the SRD or DLPF when the board was running before. When I take the power off and start the sketch again, it initializes properly.

Another thing I recognized while using your examples (I2C Basic and SPI Interrupt) is that the magnetometer values constantly turn to 0 after I moved the IMU around. I haven't looked deeper into that yet. 

Regards,
Dom

AzureSparkle, 2020-04-30:
I'm getting a similar issue, I get a lot of failures between
-15 to -20

I'm using an ESP32 and an external power supply, so I think I can limit the error to SPI problems

closed 2021-02-14 by flybrianfly