bolderflight/invensense-imu #29: calibrateAccel

LuckyCIover, 2018-06-12:
Is it works? It's every time show me that bias is 0, and scale is 1. But i can read peak 10.7mss on X and min -9.0 or another => real bias ~ -0.9. I checked sourses, but i can't find something wrong.

flybrianfly, 2018-06-12:
Yes, make sure that you are running calibrateAccel 6 times total, once in each axis (+/-). The method is using the gravity vector as input to compute scale factor, which is why this is necessary.

LuckyCIover, 2018-06-12:
Thank you so much, my English don't pretty good and I looked once again at the code and understood, that it should be run at least 6 times to calculate all biases, wrote some useful lines in your code (it helped me) to print values of measuring when pass the loop with (_numSamples).
In my opinion your library is the most understandable, easy, but powerful. Thanks.

closed 2018-06-12 by LuckyCIover