bolderflight/invensense-imu #31: Mag bias and scale returning 0

zenmanenergy, 2018-07-21:
After I run the calibrateMag() method the bias and scales are returning 0.000 and 1.000

I have tried waving it in a big figure 8, a small figure 8, a slow figure 8, fast figure 8 and leaving it still. 

Any idea why?

flybrianfly, 2018-07-21:
Is calibrateMag() returning control before you print the bias and scale factors? Is the return value from calibrateMag a positive number? Thanks.

zenmanenergy, 2018-07-21:
Uggh, I must have changed something in the code. When I rolled back to a previous version, I'm getting totally different results. I added in a MadgwickAHRS method and I must have f'd something up. I'm going to close this issue in a minute. 

While I have you here, can you explain a few things about the calibrateMag() method? I am struggling to understand how it works.

-What's the difference between bias and scale?
-What are the variables: _delta, _framedelta and _deltaThresh doing?
-What is causing this to be true: if (_framedelta > _deltaThresh) {  That seems to cause the loop to restart again.

flybrianfly, 2018-07-22:
In a perfect world with a perfect sensor, a 3D magnetometer would describe a sphere. In reality, the magnetometer will be biased (i.e. not centered) and scaled. So the algorithm is trying to find the maximum and minimum reading in each magnetometer direction. It then uses the max and min to find the bias and scale factor for each channel to try to center the data and fit it to a sphere. I use a continuous motion figure 8 for trying to find the max and min. delta is finding the difference between the new max / min and the previous max / min. frame delta is the largest magnitude of the delta's and the threshold is the value below which we're no longer finding useful data for determining the max / min in each channel.

zenmanenergy, 2018-07-23:
Is bias moving the center point of the 3 circles, i.e. xy plane, xz plane and yz plane?

And scale is averaging the radiuses of each so they go from an egg shape to a circle shape? Am I close?

zenmanenergy, 2018-07-23:
I'm playing around with a whole new algorithm at the moment. I don't know if it will work, but instead of trying to find maxs and mins, I'm creating arrays of x/y points that have similar z values (up to 3 digits at the moment).

Then I'm going to loop through each point in pairs of 2 and calculate the center of the circle they would exist on, then average all the centers on each plane and use that to calculate the radius

I *think* it should give me a line of centers. Do that for the three planes and where they intersect should be the point to calculate the bias.

It's super complicated, but I think the advantage is that you won't have to wiggle the board as much.  With the current calibration, if you never hit the max and min points the calibration will be off.

flybrianfly, 2018-07-23:
Your description of the current algorithm is correct. Your proposed approach sounds similar to a least squares approach, this is a good resource that I've seen (the description is for an accelerometer, but the approach is the same for a magnetometer):
https://chionophilous.wordpress.com/2011/08/26/accelerometer-calibration-iii-improving-accuracy-with-least-squares-and-the-gauss-newton-method/

The calibration should be better at the cost of more complexity and memory usage.

zenmanenergy, 2018-07-23:
I'm not totally following his description of the least squares model. I'll have to read it a few times before I understand. From my first read it sounds dramatically different than what I'm attempting. 

My aim is to identify the sphere through a series of sliced circles. For example, xy points that have the same Z value should have the center point (obviously have to account for noise). Noise will cause problems, but with two concentric circles we can draw a line between those centers and identify the Z axis. So in theory it should only take 4 total points to identify the Z axis. Do that for the other two planes and it essentially only takes 6 pairs of points or 12 total points to find the intersection of the 3 axis' which should calculate the bias. As long as the pairs are on the same planes it won't matter where the points are in the spheres. I haven't figured out the scale yet. One idea at a time!

This idea is still very much a work in progress, but the benefit that I see to doing it this way is you should need significantly less data. I found that with about 1 second of waving i ended up with a long list of points that were all on the same planes. 

I'm building a processing interface connected through the serial port so I can visualize what I'm trying to do.

closed 2018-07-26 by flybrianfly