bolderflight/invensense-imu #32: add lib to library manager

aster94, 2018-07-24:
hi brian, as i always say this is the best library for mpu9250
you should add it to the library manager
please see: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

idem for the AHRS lib

per1234, 2018-07-24:
This will only require three things to be done:
1. Add library.properties file (which could be done by merging https://github.com/bolderflight/MPU9250/pull/25).
1. Create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in library.properties.
1. Open an issue requesting the library to be added at the arduino/Arduino repository: https://github.com/arduino/Arduino/issues.

flybrianfly, 2018-07-26:
Thanks for the suggestion, I've completed the steps and requested this be added to the library manager. I'll submit similar requests to our other libraries.

closed 2018-07-26 by flybrianfly