bolderflight/invensense-imu #33: Calculation of Magnetometer Heading (raltive to magnetic north / 0 -360^°

wilhelmzeuschner, 2018-07-31:


SamiXSous, 2019-01-23:
@wilhelmzeuschner I am currently facing the same problem. Did you find a way to resolve this issue? If so please do share.

Thanks in advance!

flybrianfly, 2019-01-23:
Sorry I missed this. The code below will give you heading if you keep the IMU level, it's similar to the code posted by @wilhelmzeuschner. Please note that you absolutely need to calibrate the magnetometer on the MPU-9250 for this to work. The correction factors are typically very large with these sensors. The magnetometer calibration is all of the EEPROM stuff that is happening (i.e. loading calibration factors).

`
#include "MPU9250.h"
#include "EEPROM.h"

/* magnetometer data */
float h, hx, hy, hz;
/* magnetometer calibration data */
float hxb, hxs, hyb, hys, hzb, hzs;
/* euler angles */
float yaw_rad, heading_rad;
/* filtered heading */
float filtered_heading_rad;
float window_size = 20;
/* conversion radians to degrees */
const float R2D = 180.0f / PI;
/* MPU 9250 object */
MPU9250 imu(Wire, 0x68);
/* MPU 9250 data ready pin */
const uint8_t kMpu9250DrdyPin = 2;
/* Flag set to indicate MPU 9250 data is ready */
volatile bool imu_data_ready = false;

/* ISR to set data ready flag */
void data_ready()
{
  imu_data_ready = true;
}

void setup()
{
  /* Serial for displaying results */
  Serial.begin(115200);
  while(!Serial) {}
  /* 
  * Start the sensor, set the bandwidth the 10 Hz, the output data rate to
  * 50 Hz, and enable the data ready interrupt. 
  */
  imu.begin();
  imu.setDlpfBandwidth(MPU9250::DLPF_BANDWIDTH_10HZ);
  imu.setSrd(19);
  imu.enableDataReadyInterrupt();
  /*
  * Load the magnetometer calibration
  */
  uint8_t eeprom_buffer[24];
  for (unsigned int i = 0; i < sizeof(eeprom_buffer); i++ ) {
    eeprom_buffer[i] = EEPROM.read(i);
  }
  memcpy(&hxb, eeprom_buffer, sizeof(hxb));
  memcpy(&hyb, eeprom_buffer + 4, sizeof(hyb));
  memcpy(&hzb, eeprom_buffer + 8, sizeof(hzb));
  memcpy(&hxs, eeprom_buffer + 12, sizeof(hxs));
  memcpy(&hys, eeprom_buffer + 16, sizeof(hys));
  memcpy(&hzs, eeprom_buffer + 20, sizeof(hzs));
  imu.setMagCalX(hxb, hxs);
  imu.setMagCalY(hyb, hys);
  imu.setMagCalZ(hzb, hzs);
  /* Attach the data ready interrupt to the data ready ISR */
  pinMode(kMpu9250DrdyPin, INPUT);
  attachInterrupt(kMpu9250DrdyPin, data_ready, RISING);
}

void loop()
{
  if (imu_data_ready) {
    imu_data_ready = false;
    /* Read the MPU 9250 data */
    imu.readSensor();
[... 165 lines cut ...]
  Serial.print("Calibrating magnetometer, please slowly move in a figure 8 until complete...");
  imu.calibrateMag();
  Serial.println("Done!");
  Serial.print("Saving results to EEPROM...");
  /* Save to EEPROM */
  value = imu.getMagBiasX_uT();
  memcpy(eeprom_buffer, &value, sizeof(value));
  value = imu.getMagBiasY_uT();
  memcpy(eeprom_buffer + 4, &value, sizeof(value));
  value = imu.getMagBiasZ_uT();
  memcpy(eeprom_buffer + 8, &value, sizeof(value));
  value = imu.getMagScaleFactorX();
  memcpy(eeprom_buffer + 12, &value, sizeof(value));
  value = imu.getMagScaleFactorY();
  memcpy(eeprom_buffer + 16, &value, sizeof(value));
  value = imu.getMagScaleFactorZ();
  memcpy(eeprom_buffer + 20, &value, sizeof(value));
  for (unsigned int i = 0; i < sizeof(eeprom_buffer); i++) {
    EEPROM.write(i, eeprom_buffer[i]);
  }
  Serial.println("Done! You may power off your board.");
}

void loop() {}
`

closed 2019-01-23 by flybrianfly

AndrewNoviello, 2021-01-01:
Hi,
I am attempting to use this library for robotics using the MPU9250. I am using your code above, but am getting strange and inconsistent values. First of all, after the first few seconds of running, the sensor values stop changing and all the values stay the same from then after. Do you have any ideas as to why? Secondly, even when rotating it around, the sensor gives the same output for multiple different rotation spots, all in total opposite orientations. Could this be a calibration issue? Thanks for any help!

YashviShahBvn, 2024-11-16:
> ```c++
> pitch_rad = asinf(ax);
> ```

In this, the pitch calculation is done for relatively small pitch values. What I found is that pitch is calculated using this formula: `pitch = atan2(-ax, sqrtf((ay*ay) + (az*az)))`
I want to understand the difference between both formulae and for real life autonomous applications, which is most appropriate formula?