bolderflight/invensense-imu #34: Reset SRD in calibration methods

antonays, 2018-08-06:
Hi,

Thanks for the library! Well done and very robust and easy to debug code.

I am having the following problem regarding the Sensor Calibration routines, all of them (calibrateGyro(), calibrateAccel(), calibrateMag()).
In the beginning of each of them, we set the SRD to be 19 such that we have a frequency of 50 Hz, or a sample coming every 20ms.
My understanding is that this is required to perform correctly the sensor setup.

However, In the end of each of the calibration methods, we try to set the SRD back to the previous value, the one known as the protected _srd value.
For me, trying to do this always fails, as if the SRD method doesn't work when the SRD is set to 19.
Following this problem, the behavior i am observing is that if i run one of the calibration methods, i stay at an SRD of 19, regardless of the value i actually set, and the setup fails (returning a negative number etc'..)

Commenting the instructions to set the SRD to 19, everything works fine and it looks like the sensor is calibrated and generates stable values.

My setup is:
I am using an Arduino Nano for the MCU to interface the sensors. And a non-latching interrupt based sampling, such that an ISR is waiting for an interrupt and then instructs the MCU to go and fetch the data.
Everything works well until i try to change the calibration.

Any help would be appreciated.
Thank you!

flybrianfly, 2018-08-10:
Thanks! That's a known issue #13 that I haven't gotten around to yet. In the meantime, I would suggest using the setSrd method to set it back after calibration.

Ananfruit, 2020-10-26:
Can you share the code when you calibrate your accel and gyro, please? Thanks in advance!

closed 2021-02-14 by flybrianfly