bolderflight/invensense-imu #35: Using this library on MPU6050

aliaboshady123, 2018-08-23:
I tried using this MPU9250 library on my MPU6050 because I found this library to be simpler than the ones used for the MPU6050. However it didn't work, though some people said it would.

So, is there anyway to use this library on MPU6050?
Thank you

flybrianfly, 2018-08-23:
It would take a lot of modification to work; although, it's fairly well commented, so should be doable. You would have to remove all of the magnetometer code in _begin_, _readSensor_, and some of the configuration methods (definitely _setSRD_, but I don't remember if any of the others are interacting with the magnetometers). In _begin_ there is a WHOAMI check, you would need to change the allowable value to the 6050 WHOAMI return value.

aliaboshady123, 2018-08-24:
Thanks for replying.
And wow this seems like a lot of work looking at the code. You're not planning on making a version for MPU6050 by any chance, are you? XD

closed 2019-01-26 by flybrianfly