bolderflight/invensense-imu #36: MPU9250 x and y axis are reversed?

larryyyyy, 2018-08-31:
I found that the data I get by simple importing the basic_i2c.ino and the data I got is not in a correct order.  
Ax and Ay are reversed and gx and gy as well. does anyone get the same problem?
I am using this chip: https://www.makerlab-electronics.com/product/mpu9250-9-dof-3-axis-accelerometer-gyro-magnetometer/

larryyyyy, 2018-08-31:
just found that the default oriantation matrix is 
const int16_t tX[3] = {0,  1,  0};
const int16_t tY[3] = {1,  0,  0};
const int16_t tZ[3] = {0,  0, -1};

wondering why is setting like this? shouldn't it be:
const int16_t tX[3] = {1, 0, 0};
const int16_t tY[3] = {0, 1, 0};
const int16_t tZ[3] = {0, 0, 1};
?

Thank you!

flybrianfly, 2018-08-31:
By default the axis used for the MPU-9250 accel and gyro data is different from the magnetometer data. This library transforms all of the data to a common axis system, which also is a standard aircraft axis system (x out the nose, y to the right, z down) and is described here:
https://github.com/bolderflight/MPU9250/blob/master/README.md#sensor-orientation

closed 2018-08-31 by flybrianfly

larryyyyy, 2018-08-31:
I see! Thank you so much! 
However, what if I who to have the really raw data for demonstration, is that I just simply set the orientation to the following? Thank you
const int16_t tX[3] = {1, 0, 0};
const int16_t tY[3] = {0, 1, 0};
const int16_t tZ[3] = {0, 0, 1};

flybrianfly, 2018-09-04:
Yes, it should be that simple.