bolderflight/invensense-imu #38: Basic_I2C Example with ESP32

only0netom, 2018-10-25:
Hey, I'm trying to use the MPU9250 in combination with an ESP32 Dev Board instead of an Arduino. I already tested the connection with an I2C Scanner and it shows that there is a device connected with address 0x68 (SDA and SCL are Pins 21 and 22 on the ESP32). 
But when I start the Basic_I2C Example, I always get status = -1 from IMU.begin(). My wiring should be correct. 
Do you have any idea what the problem is and how I can get this working?

flybrianfly, 2018-10-25:
Could it be the same issue as this?
https://github.com/bolderflight/BME280/issues/2

I think something might be up with the two wire implementation on ESP32.

flybrianfly, 2018-10-25:
Also might be related to:
https://github.com/espressif/arduino-esp32/issues/1998

imalipusram, 2018-10-27:
I think it's not an ESP-32 bug. The BME280 library at https://github.com/finitespace/BME280 is communicating with the sensor and displaying reasonable data.

flybrianfly, 2018-10-27:
That library does not use a repeated start for its I2C communication. Both my MPU-9250 and BME-280 libraries use repeated starts and the ESP-32 bug is about repeated starts. As a test, you could search for 'endTransmission(false)' and replace with 'endTransmission()'. I think it only occurs in the readRegisters method.

only0netom, 2018-10-28:
Thanks for the replies. If I replace 'endTransmission(false)', the I2C communication works. I also managed to get it working with SPI. 

But I have to comment out several parts of the begin function to get it working:
- the whoAmI byte is 112 instead of 113, so I get **status = -5** if I don't comment this out
- the whoAmIAK8963 byte is 0, so the begin function returns **status = -14**
- and I also get **status -16, -18 and -20**, which means I don't get values from the magnetometer and I can't calibrate the gyro.

Is this also a ESP32 related problem?

flybrianfly, 2018-10-28:
No, this is a problem with your sensor - the WHO AM I value that you're reporting indicates that your sensor is a MPU-6500, not an MPU-9250. Which means it does not have a magnetometer (why my code is reporting a magnetometer WHO AM I of 0 and is unable to communicate with it).

Celppu, 2019-01-04:
'endTransmission(false)' must be changed to 'endTransmission()' for esp32 !! I had to only change that to get it working on esp32. Before I got error -1 . Row 1025 on MPU9250.cpp  .

closed 2021-02-15 by flybrianfly

sam17012k1, 2021-06-04:
> Thanks for the replies. If I replace 'endTransmission(false)', the I2C communication works. I also managed to get it working with SPI.
>
> But I have to comment out several parts of the begin function [...]

Hi i am also facing the same issue, Could you please describe in detail with what helped you solve this

flybrianfly, 2021-06-04:
Replace endTransmission(false) with endTransmission() throughout the code.