bolderflight/invensense-imu #39: Accelerometer and Gyroscope readings are reversed.

nickgoza, 2018-11-06:
Hello, 

Thank you all for the information on this topic.

I have modified the advanced I2C to add a filter to stabilize the data and wrote test code to graph each axis (one program per axis per reading) in real time for testing. I've found that my Accelerometer readings are reading the Gyro (it is still displaying with the Accelerometer units) and my Gyro is reading acceleration in radians. 

My functions are correct (I did not put the wrong one in). I am using an Arduino Uno with a sparkfun MPU-9250. I believe this has something to do with the header file, but I am not experienced enough to identify what I would need to interchange. Can I get some help on this?

Let me know what I can provide to be helpful.

Thank you, 
Nick

nickgoza, 2018-11-06:
As an afterthought, it might be helpful to mention this applies to the X,Y, and Z axis on both

flybrianfly, 2018-11-07:
Hi Nick,

Can you attach your code?

Thanks,
Brian

nickgoza, 2018-11-07:
Wow, everything makes sense now...I misunderstood what the sensors did...haha. I am new to this. Thank you for your help and your code has been very helpful to me!

Nick