bolderflight/invensense-imu #42: ACCEL_RANGE is ignored

maikischa, 2018-12-31:
When I set the Accel Range with `IMU.setAccelRange(MPU9250::ACCEL_RANGE_4G);` the statement is ignored and all readings I get are limited to 20 m/s/s (2G).
I have tried different ranges but always get only maximum 2G readings.

maikischa, 2018-12-31:
In addition I found out that the ACCEL_RANGE is only rested to 2G if I run `IMU.calibrateAccel()`.

It seems that `IMU.calibrateAccel()` resets the ACCEL_RANGE to 2G

ChisholmKyle, 2020-10-14:
I noticed in the code the calibrate functions do not properly restore settings after changing them for the calibration. I made a fix in pull request #76 which should actually restore settings.

flybrianfly, 2021-02-15:
Removed calibration from the library, the accel range should be set correctly now.

closed 2021-02-15 by flybrianfly