bolderflight/invensense-imu #44: Accelometer Z axis

Mika857, 2019-01-20:
The Z axis is not changing its value but I turn it at the right position and it is only changing if I turn it at the X or the Y axis then its the opposit of the x/y value. Do you know something about this problem? 
(The gyroscop and the magnetometer are working correctly)

flybrianfly, 2019-01-20:
If the Z axis is pointed up or down, you'll read -/+ 9.87 m/s/s. If the X axis or Y axis is pointed up or down, Z will measure 0.

Mika857, 2019-01-22:
[image]

In this position the Z measure -10 and if I turn it like the arrows or in the another direction the Z is not changing anyway but I need this to let the robot turn correctly. Can you help me?

flybrianfly, 2019-01-22:
No, I can't see the image. But you can't use accelerometers to measure heading - only pitch and roll. You'll need to use the magnetometers to measure heading.

Mika857, 2019-01-22:
Ohh ok didnt knew that.
Thank you for helping me.
But I saw videos where people used the accelometer for measuring the heading how is this then possible?

Mika857, 2019-01-26:
ok now I tried it with the magnetometer to measure heading but new I have now other problems.

If I turn it to the left the x value goes from 0 to 20 and back but if I turn it to the right it goes from 0 to only 14(less / more).

Is this correct and why does the x value measures heading and not the z?

Mika857, 2019-01-26:
is there a way of calibrating the mpu9250

Mika857, 2019-01-26:
and is there really no way to use the accelometer for heading? Cause I need this for turning a robot correctely and the magnetometer is a little bit a problem because if I hold my smartphone near to it the value is completely different than before.

flybrianfly, 2019-01-26:
See the calibration methods in this library for calibrating the magnetometer. It needs to be calibrated for it to work properly. Also, any time you change its magnetic environment, you would need to re-do the calibration. Finally, the magnetometer is quite noisy, I recommend a first order IIR filter to smooth the output for use as a compass.

Gravity, which is what the accelerometer is measuring in steady conditions, only acts vertically. There is no way to get heading information from it.

closed 2019-01-26 by flybrianfly