bolderflight/invensense-imu #45: 2x MPU9250 i2c Esp32

imaniroman, 2019-03-03:
Hey Guys,
I am having issues with my esp32 and interfacing my two IMUs (MPU9250 with different addresses). Both Imus are connected to 21,22 pins of the Esp32. My setup works at times and then it doesn't work. Here's the story -

2 months ago - Got Everything working. Then it stopped. Replaced the IMUs, eventually after many resets, it began to work. Both sets of IMUs worked fine.
AND KEPT WORKING
1 week ago - IMU's stopped work (Previous night everything worked fine!!)

I am using a basic i2c scanner sketch to test my communication. When both IMUs are connected. I am getting 'no devices found 'errors. When I connect them alone, both addresses are picked up just fine. Tried the basic I2c sketches. I get the -1 status errors.

I am not using any pullup resistors. As I believe, they come with the breakout boards (generic gy boards). I don't think I need them as my setup used to worked fine for months!!. And If I do use pull resistors, what would the circuit look like given that two IMU's are being used, will the resistors be in parallel (Electronics noob here)

PLEASE HELP as these IMUs are essential to my thesis due soon!

74ls04, 2019-03-03:
The answer to this issue on this other library should apply to your case https://github.com/kriswiner/MPU9250/issues/37

imaniroman, 2019-03-04:
@74ls04 Thank you for the reply. I am not sure where in the post lies my answer, Sorry that I am a bit slow. How do I tweak my I2C scanner to get both addresses? I am currently using this scanner to check for the devices..

I've added Wire1.begin(5,4) and  Wire1.beginTransmission(address);. I am trying to use the second I2c on a different set of pins to see if I get better results.. still no luck, only able to find one imu attached to 21,21 pins

```
#include <Wire.h>
void setup()
{
  Wire.begin(21,22);
  Wire1.begin(5,4);
 
  Serial.begin(9600);
  while (!Serial);             // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner");
}
 
 
void loop()
{
  byte error, address;
  int nDevices;
 
  Serial.println("Scanning...");
 
  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    Wire1.beginTransmission(address);
    error = Wire.endTransmission();
 
    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");
 
      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
 
  delay(500);           // wait 5 seconds for next scan
}
```

74ls04, 2019-03-06:
Have you addressed the IMUs by connecting the AD0 pin of one to ground and the AD0 pin of the other one to VCC with a resistor? If you've done that then they will each have unique addresses and you shouldn't have any issues. Here is the relevant section from the [documentation](http://43zrtwysvxb2gf29r5o0athu.wpengine.netdna-cdn.com/wp-content/uploads/2015/02/MPU-9250-Datasheet.pdf)
[image]

You should also add pull-up resistors to VCC on the SDA/SCL lines.

imaniroman, 2019-03-06:
@74ls04 Thanks again for the reply, I did not use a resistor, but I have connected one of the AD0 pins to VCC to get 69 as the address. While the other, stayed unmodified with the default address of 68. The addresses can be checked using the I2C scanner separately. I did not use a resistor.

I am having the same issues with an arduino pro mini (just to see if it's an issue with the Esp32). I had both the imus working for about 10 seconds. Both being connected to the SDA and SCL pins of the pro mini. Screenshot of the imus being found is below.

[image]

74ls04, 2019-03-07:
Try adding pull-up resistors on the I2C lines -- in general they will improve stability of the bus.

imaniroman, 2019-03-27:
Partially resolved the issue just by using 2 Pro minis to monitor the 2 Imus.. Data from pro minis was sent to Esp using serial. Thank  you everyone for your support!

closed 2019-03-27 by imaniroman