bolderflight/invensense-imu #49: Data Recording and Storage at 1Khz (New to Arduino)

Yantobanto, 2019-04-30:
Hi there, 

I have an issue with a project i'm working on. 

I have an MPU-9250 and SD Card Module  (With a compatible SanDisk 16GB SD Card) connected to an official Arduino UNO board.

I want to be able to record and store the data at a 1KHz sample rate. I believe two issues are perhaps the limitation of bolderflight package, as well as myself using print to read/write to the data, and not storing binary and using a buffer. If anyone could look into this that would be amazing, I'll be very active in checking this thread.

Thank you,

I've attached the file of code I've currently been using. 
[Arduino Example Issue.pdf](https://github.com/bolderflight/MPU9250/files/3131317/Arduino.Example.Issue.pdf)

Yantobanto, 2019-04-30:
``
#include "MPU9250v2.h"
#include <SD.h>

// an MPU9250 object with the MPU-9250 sensor on I2C bus 0 with address 0x68
MPU9250 IMU(Wire,0x68);
int status;
const int chipSelect = 4;

void setup() {
  // serial to display data
  Serial.begin(115200);
  SD.begin(chipSelect);
  
  // while(!Serial) {}

  // start communication with IMU 
  status = IMU.begin();

  // setting the accelerometer full scale range to +/-8G 
  IMU.setAccelRange(MPU9250::ACCEL_RANGE_8G);
  // setting the gyroscope full scale range to +/-500 deg/s
  IMU.setGyroRange(MPU9250::GYRO_RANGE_500DPS);
  // setting DLPF bandwidth to 20 Hz
  IMU.setDlpfBandwidth(MPU9250::DLPF_BANDWIDTH_460HZ);
  IMU.setSrd(0);
}

void loop() {
  
  IMU.readSensor();
  String dataString = "";
  dataString += String(IMU.getAccelX_mss(),2);
  dataString += ",";
  dataString += String(IMU.getAccelY_mss(),2);
  dataString += ",";
  dataString += String(IMU.getAccelZ_mss(),2);

  File dataFile = SD.open("datalog.txt", FILE_WRITE);
  dataFile.println(dataString);
  dataFile.close();
  Serial.println(dataString);
  
  
  
  delay(0);
}

licesonw, 2020-11-06:
@Yantobanto did you figure it out? I would also be interested in a solution.

paulrossmanith, 2021-02-06:
Hey There,

I did some testing myself with an MPU9250, an Arduino Mega and an SPI-SD-Card board.
I found writing to the SD-card limits the polling rate greatly. When I send the Data over Serial to my PC, I get about 1300 Hz (~ 0.8 ms per loop) when using 1 MBaud (includes full set of raw data, 10 floats). Using the SD-Card, i can expect to get around 50-60 Hz (~ 166 ms per loop).
While the speed of the SD-Card will matter (mine is pretty slow), I do believe you will always be limited by the SPI-Speed.

#### Let's do a rough estimation
Assume your SD-Card can do about 10 MB/s write. SPI will do a maximum of 1 MBit/s or 125 kB/s (keep in mind, this is not the net-datarate!) Your dataset includes 11 floats per write á 4 Byte (Temperature, Timestamp in Millis and 9x Sensor readings), so each dataset is 44 Byte.

(125 * 1024 Byte/s) / 44 Byte = 2.909,1 1/s = **2.909,1 Hz.**
This is the maximum you can expect. This **does not account for** additional overhead caused by write-errors, idle-times, the time your Arduino can not write to your SD-Card because it is reading the I2C-bus, the speed and overhead of the SD-Cards library and so on...

#### What I did
If you really need to record your data at such a high speed, consider using a PC with a Serial Logger implemented in Python or C++ (just google it). For even higher Datarates with more performance headroom, get yourself a Teensy or an Arduino Due.
You can increase your Serial Baudrate up to 500000 bauds with the Mega without compromising the I2C-reads. A Teensy will go up to 4800000 bauds with no issues.

closed 2021-02-15 by flybrianfly