bolderflight/invensense-imu #51: WHOAMI byte

Colidar, 2019-05-14:
Hi all,

i am using MPU9250 with an arduino nano and arduino IDE 1.8.9.
When I'm trying to use the BASIC_I2C sketch there is an error with status -5...

IMU initialization unsuccessful
Check IMU wiring or try cycling power
Status: -5

I have checked the error in the MPU9250.cpp:
 // check the WHO AM I byte, expected value is 0x71 (decimal 113) or 0x73 (decimal 115)
  if((whoAmI() != 113)&&(whoAmI() != 115)){
    return -5;
  }

But i don't know how to deal with this error and to run the sketch properly..
I'm happy about your advice!

flybrianfly, 2019-05-14:
Sounds like it isn't actually an MPU-9250 IMU. You can add the following right before the return statement and we can look up the sensor you're using:

`
Serial.println(whoAmI());
`

Colidar, 2019-05-15:
thanks for your answer!
I'm pretty sure that it's a MPU-9250 IMU..
[image]

My WHO AM I value is 112 (0x70 in hex).. 
So i changed it in the MPU9250.cpp file but then i got the error -14. 
Once again i print my value and it 0 instead of expected 72.. 

Any new advice?

flybrianfly, 2019-05-15:
The WHO_AM_I value indicates that you have an MPU-6500. This is only a 6 axis IMU and does not have the magnetometers, which is why it then starts failing at the magnetometer initialization. You could modify the library to remove all of the magnetometer calls and it should mostly work.

closed 2019-05-15 by flybrianfly

Colidar, 2019-05-15:
nice to know :D thanks a lot, it has to be a failure in delivery