bolderflight/invensense-imu #52: Accelerometer calibration

Lynxu, 2019-06-08:
When i try to calibrate the accelerometer i call the function for each axis and when i show the results all of the biases are 0 and the scale for each one is 1. Is that normal?

`CALIB Gyro -- dont move until I say stop!!! done!
X: 0.004778
Y: 0.019271
Z: -0.003822
Gyro calib done
********** ACC calib **************
Enter when ready for dir 1 X+..
Enter when ready for dir 2 X-..
Enter when ready for dir 3 Y+..
Enter when ready for dir 4 Y-..
Enter when ready for dir 5 Z+..
Enter when ready for dir 6 Z-..
Acc calib done
Vals:
X: 0.000000/1.000000 (bias/scale)
Y: 0.000000/1.000000 (bias/scale)
Z: 0.000000/1.000000 (bias/scale)
CALIB MAG -- move in figure 8s until I say stop!!! done!
X: 28.451765/0.883743
Y: 47.327587/1.067171
Z: -51.208195/1.073661`

Code used:

`Serial.println(F("********** ACC calib **************"));

  char * dirs[6] = { "X+", "X-", "Y+", "Y-", "Z+", "Z-"};

  for (uint8_t i = 0; i < 6; i++) {
    Serial.print(F("Enter when ready for dir "));
    Serial.print((int)(i + 1));
    Serial.print(' ');
    Serial.print(dirs[i]);
    while (! Serial.available() ) {
      delay(10);
    }

    while (Serial.available()) {
      Serial.read();
      delay(5);
      Serial.print('.');
    }
    Serial.println();
    IMU.calibrateAccel();
  }

  Serial.println(F("Acc calib done"));

  Serial.println(F("Vals: "));

  Serial.print(F("X: "));
  Serial.print(IMU.getAccelBiasX_mss(), 6);
  Serial.print('/');
  Serial.println(IMU.getAccelScaleFactorX(), 6);

  Serial.print(F("Y: "));
  Serial.print(IMU.getAccelBiasY_mss(), 6);
  Serial.print('/');
  Serial.println(IMU.getAccelScaleFactorY(), 6);

  Serial.print(F("Z: "));
  Serial.print(IMU.getAccelBiasZ_mss(), 6);
  Serial.print('/');
  Serial.println(IMU.getAccelScaleFactorZ(), 6);`

zerru, 2019-07-07:
I have the same issue.

Any suggestion?

Thanks!

arizzadamm, 2019-07-21:
hey, can u post all of code you used to  calibration all of sensor?

arizzadamm, 2019-07-21:
@Lynxu hey, can u post all of code you used to calibration all of sensor?

Ananfruit, 2020-10-26:
Can you share the code when you calibrate your accel and gyro, please? Thanks in advance!

flybrianfly, 2021-02-15:
Not normal, it doesn't sound like the calibration was working, maybe the sensor wasn't being moved? In any case, the calibration functionality has been removed from this library.

closed 2021-02-15 by flybrianfly