bolderflight/invensense-imu #53: Magnetometer values freeze after some time

amoghskulkarni, 2019-06-09:
Hello all,

**Setting**:

I am working on a system where multiple MPU9250 IMUs (6, to be specific) are connected to an ESP32 MCU via _SPI_. Using SPI as the communication bus is crucial because the I2C pins are occupied. If it's relevant, sensors are divided as 3 sensors to VSPI and 3 sensors to HSPI. 

**Problem**: 

The Mag values freeze for multiple sensors after being initialized without any problem. The sketch follows the initialization sequence given in the example (`Basic_SPI`)[https://github.com/bolderflight/MPU9250/blob/master/examples/Basic_SPI/Basic_SPI.ino] from the examples directory. Moreover, freezing of the mag values is not deterministic between different runs, proving it very hard to detect what is going on. I have tried keeping every sensor steady (i.e. without any motion, so that the SPI connections remain stable), but the problem persists. Also - Accel and Gyro continue to give the data as expected, so the SPI connection between the MCU and IMUs cannot be an issue.

Can you please provide some pointers as to how should I try debugging this behavior of the sensors?

Moosdijk, 2019-09-24:
I'm having the same problem. It's usually after a few seconds on my end. 
I've tried playing with the SRD, but to no avail.

Did you find a fix for this issue?

amoghskulkarni, 2019-10-09:
Hi @Moosdijk,

Thanks for taking the time to reply. I am still facing the same issue, and I had almost given up on the mag. 

Can you please let me know which sensor does the mag work for? My sensors are fairly new (bought them about 3 months ago). Do you mean a newer version of MPU9250 or ICM-20948?

Moosdijk, 2019-10-09:
@amoghskulkarni Sure thing! I've also bought three sensors when I started my project. The ones I bought were the cheap ones you can get everywhere. [These](https://www.tinytronics.nl/shop/nl/sensoren/mpu-9250-accelerometer-gyroscope-magnetometer-9dof-module-3.3v-5v) were the ones I bought.

I'd like to try the ICM-20948 though!

amoghskulkarni, 2019-10-09:
Thanks for the info! I will try to find the sensors from the same manufacturer here in the US.

closed 2021-02-15 by flybrianfly