bolderflight/invensense-imu #54: Not working at all for me I2C

iFrostizz, 2019-07-01:
Arduino --> MPU9250
5V --> VCC
GND --> GND
A4 --> SDA
A5 --> SCL

IMU initialization unsuccessful
Check IMU wiring or try cycling power
Status: -1

Sensor recognized in the I2C scanner

Willus8888, 2019-07-01:
@iFrostizz what model of Arduino?

iFrostizz, 2019-07-01:
Nano V3

iFrostizz, 2019-07-01:
Don't work on Arduino Uno too

flybrianfly, 2019-07-01:
Are you using pullup resistors? What address is your I2C scan showing?

iFrostizz, 2019-07-01:
I'm not using any resistor, shall I ?
Got the 0x68 address on the I2C scanner

iFrostizz, 2019-07-06:
My MPU isn't recognised by any library

Doom4535, 2019-07-06:
@iFrostizz 
Is this issue resolved?  If not, what IMU board are you using (some have on-board pull up resistors, you have to have pull up resistors for I2C to work).

iFrostizz, 2019-07-06:
Not solved yet
I have a MPU9250 but I can read from gyro (only x and y) and accelerometer (x, y, z) using registers

Doom4535, 2019-07-06:
So, you can read from the MPU9250 without the library, but when you use the library it doesn't work?  That sounds odd, are you getting valid data when you talk straight to the MPU9250 with your code and what is your exact setup?

Can you past your working and not working code along with a picture of your setup?

iFrostizz, 2019-07-06:
I know ahah
The only way to read data is to plug my sensor to 3,3V (I forgot if it works also in 5V)
The data seems valid. I couldn't confirm for accelerometer, but I mapped raw gyro data to degrees and it seems pretty nice.

I'm sorry, I don't have my code since I don't have my pc right now, I'll provide you my code later

iFrostizz, 2019-07-10:
Here is the working code:

#include <Wire.h>

#define MPU9250_ADDRESS 0x68
 
#define GYRO_FULL_SCALE_1000_DPS 0x10
 
#define ACC_FULL_SCALE_4_G 0x08

float gx, gy, gz, ax, ay, az;

void I2Cread(uint8_t Address, uint8_t Register, uint8_t Nbytes, uint8_t* Data)
{
// Set register address
Wire.beginTransmission(Address);
Wire.write(Register);
Wire.endTransmission();
 
// Read Nbytes
Wire.requestFrom(Address, Nbytes); 
uint8_t index=0;
while (Wire.available())
Data[index++]=Wire.read();
}
 
 
// Write a byte (Data) in device (Address) at register (Register)
void I2CwriteByte(uint8_t Address, uint8_t Register, uint8_t Data)
{
// Set register address
Wire.beginTransmission(Address);
Wire.write(Register);
Wire.write(Data);
Wire.endTransmission();
}
 
void setup()
{
Wire.begin();

I2CwriteByte(MPU9250_ADDRESS,29,0x06); // Set accelerometers low pass filter at 5Hz
I2CwriteByte(MPU9250_ADDRESS,26,0x06); // Set gyroscope low pass filter at 5Hz

I2CwriteByte(MPU9250_ADDRESS,27,GYRO_FULL_SCALE_1000_DPS); // Configure gyroscope range
I2CwriteByte(MPU9250_ADDRESS,28,ACC_FULL_SCALE_4_G); // Configure accelerometers range
I2CwriteByte(MPU9250_ADDRESS,0x37,0x02); // Set by pass mode for the magnetometer
}

void loop()
{

uint8_t Buf[14];
I2Cread(MPU9250_ADDRESS,0x3B,14,Buf);

// Gyroscope
gx= (-(Buf[2]<<8 | Buf[3])) / 131.0;
gy= (-(Buf[0]<<8 | Buf[1])) / 131.0;
gz= (Buf[4]<<8 | Buf[5]) / 131.0;

// Accelerometer
ax= (-(Buf[8]<<8 | Buf[9])) / 16384.0;
ay= (-(Buf[10]<<8 | Buf[11])) / 16384.0;
az= (Buf[12]<<8 | Buf[13]) / 16384.0;

gx = constrain(gx, -63.0, 63.0);
gy = constrain(gy, -63.0, 63.0);
gz = constrain(gz, -63.0, 63.0);

gx = (gx*90.0)/63.0;
gy = (gy*90.0)/63.0;
gz = (gz*90.0)/63.0;
Serial.println (gx); 
Serial.print ("   ");
Serial.print (gy);
Serial.print ("   ");
Serial.print (gy);
Serial.print ("    |    ");

// Accelerometer
Serial.print (ax); 
Serial.print ("   ");
Serial.print (ay);
Serial.print ("   ");
Serial.print (az); 
Serial.println ("    |    ");

delay(100);

}

iFrostizz, 2019-07-11:
Do you have any idea ?

Doom4535, 2019-07-21:
Have you tried the [basic I2C](https://github.com/bolderflight/MPU9250/blob/master/examples/Basic_I2C/Basic_I2C.ino) example?  The library wraps the reads and writes inside readSensor().

Does this code build for you (was it copied and pasted in?), as I suspect this: `gx = (gx90.0)/63.0;
gy = (gy90.0)/63.0;` won't build (looks to be missing a multiply between the variable and 90).

Is this using an Arduino Nano or Teensy board?  Could you also provide a link to where you got your MPU9250 board (do you know the size of the pull up resistors on it)?

Your code doesn't specify the I2C clock speed, could you try setting it using `[Wire.setClock()](https://www.arduino.cc/en/Reference/WireSetClock)` to `400000` (400K).  I believe Arduino defaults to 100K, this library here tries to run at 400K.  I would then test this library with the [`_i2cRate`](https://github.com/bolderflight/MPU9250/blob/38b96cb19bccc9d45794885c891be169d0b85ea7/src/MPU9250.h#L124) variable turned down to 100K.  If these changes make a difference, you probably need to swap your pull up resistors [this is a useful article on I2C resistor sizing](http://dsscircuits.com/articles/effects-of-varying-i2c-pull-up-resistors).  If this is a homemade bare board, you have to have I2C pull up resistors for the protocol to work.

iFrostizz, 2019-07-21:
Yes, tried it but doesn't work. I have a cheap Chinese module:
https://s.click.aliexpress.com/e/UytSxxK

I bought a more expensive one but didn't received

I wrote this code but it's not really precise ... Just to visualise

Can't try the I2C clock now, I'll try it back home

iFrostizz, 2019-07-21:
For the multiplication: It's a fail in the copy/paste

iFrostizz, 2019-07-30:
Just got a new MPU board
Now, I can sometimes read values, but if I move a little bit the wires, it stops

MinayaKarimova97, 2021-02-05:
I also met the issue

flybrianfly, 2021-02-15:
For these sort of issues, I'll need information on the microcontroller used, the sensor breakout, and the wiring connections between the microcontroller and the sensor.

MinayaKarimova97, 2021-02-15:
Thanks for your replying. I changed the sensor and that one worked for me.

MinayaKarimova97, 2021-02-15:
i was using Arduino Mega, and the sensor mpu9250/6500 as his. Since i replaced it with another one, it worked

closed 2021-02-15 by flybrianfly