bolderflight/invensense-imu #55: calibrateAccel() not calculating bias/scale?

Aerugo2272, 2019-07-30:
I've been trying to run the calibration code for the MPU9250, and the calibrateMag() method seems to work, but the calibrateAccel() method simply returns bias 0 and scale 1.
I am running the function 6 times, changing the orientation of the imu with +z, -z, +y, -y, +x, -x pointing up in that order, switching whenever the "switch" serial print appears and holding position until the next.
Any ideas?

Code I'm running:
`
/*
CalibrateMPU9250.ino
Brian R Taylor
brian.taylor@bolderflight.com
Copyright (c) 2017 Bolder Flight Systems
Permission is hereby granted, free of charge, to any person obtaining a copy of this software 
and associated documentation files (the "Software"), to deal in the Software without restriction, 
including without limitation the rights to use, copy, modify, merge, publish, distribute, 
sublicense, and/or sell copies of the Software, and to permit persons to whom the Software is 
furnished to do so, subject to the following conditions:
The above copyright notice and this permission notice shall be included in all copies or 
substantial portions of the Software.
THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING 
BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND 
NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, 
DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, 
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
*/

#include "MPU9250.h"
#include "EEPROM.h"

// an MPU-9250 object on SPI bus 0 with chip select 10
MPU9250 Imu(Wire, 0x68);
int status;

// EEPROM buffer and variable to save accel and mag bias 
// and scale factors
uint8_t EepromBuffer[48];
float value;

void setup() {
  // serial to display instructions
  Serial.begin(57600);
  while(!Serial) {}
  // start communication with IMU 
  status = Imu.begin();
  if (status < 0) {
    Serial.println("IMU initialization unsuccessful");
    Serial.println("Check IMU wiring or try cycling power");
    Serial.print("Status: ");
    Serial.println(status);
    while(1) {}
  }
  // calibrating accelerometer
  Serial.println("Starting Accelerometer Calibration");
  Imu.calibrateAccel();
  Serial.println("Switch");
  delay(5000);
  Imu.calibrateAccel();
  Serial.println("Switch");
  delay(5000);
  Imu.calibrateAccel();
  Serial.println("Switch");
  delay(5000);
  Imu.calibrateAccel();
  Serial.println("Switch");
  delay(5000);
  Imu.calibrateAccel();
  Serial.println("Switch");
  delay(5000);
  Imu.calibrateAccel();
  Serial.println("Done");
  Serial.println("Starting Magnetometer Calibration");
  delay(5000);
  // calibrating magnetometer
  Imu.calibrateMag();
  // saving to EEPROM
  value = Imu.getAccelBiasX_mss();
  memcpy(EepromBuffer,&value,4);
  Serial.print(value);
  Serial.println(": abx");
  value = Imu.getAccelScaleFactorX();
  memcpy(EepromBuffer+4,&value,4);
  Serial.print(value);
  Serial.println(": asx");
  value = Imu.getAccelBiasY_mss();
  memcpy(EepromBuffer+8,&value,4);
  Serial.print(value);
  Serial.println(": aby");
  value = Imu.getAccelScaleFactorY();
  memcpy(EepromBuffer+12,&value,4);
  Serial.print(value);
  Serial.println(": asy");
  value = Imu.getAccelBiasZ_mss();
  memcpy(EepromBuffer+16,&value,4);
  Serial.print(value);
  Serial.println(": abz");
  value = Imu.getAccelScaleFactorZ();
  memcpy(EepromBuffer+20,&value,4);
  Serial.print(value);
  Serial.println(": asz");
  value = Imu.getMagBiasX_uT();
  memcpy(EepromBuffer+24,&value,4);
  Serial.print(value);
  Serial.println(": mbx");
  value = Imu.getMagScaleFactorX();
  memcpy(EepromBuffer+28,&value,4);
  Serial.print(value);
  Serial.println(": msx");
  value = Imu.getMagBiasY_uT();
  memcpy(EepromBuffer+32,&value,4);
  Serial.print(value);
  Serial.println(": mby");
  value = Imu.getMagScaleFactorY();
  memcpy(EepromBuffer+36,&value,4);
  Serial.print(value);
  Serial.println(": msy");
  value = Imu.getMagBiasZ_uT();
  memcpy(EepromBuffer+40,&value,4);
  Serial.print(value);
  Serial.println(": mbz");
  value = Imu.getMagScaleFactorZ();
  memcpy(EepromBuffer+44,&value,4);
  Serial.print(value);
  Serial.println(": msz");
  for (size_t i=0; i < sizeof(EepromBuffer); i++) {
    EEPROM.write(i,EepromBuffer[i]);
  }
  Serial.println("Done");
}

void loop() {}`

Aerugo2272, 2019-07-30:
I realise the EEPROM parts are irrelevant to me, but I have tried without those lines and the problem persists, while the IMU is clearly able to get readings judging from what I'm getting with Imu.getAccelX_mss() etc (which are returning values over +/- 9.0 in the respective positions)

nmk456, 2019-08-06:
I'm having the same issue. I have the MPU9250 connected to an ESP32 with I2C. My calibration code is pretty much the same as yours, and I get the same result of bias 0 and scale 1 for every axis.

anst270125, 2019-10-07:
I have the same issue.

anst270125, 2019-10-07:
@Aerugo2272 @nmk456 I found the problem. All the abs() calls in the calibrateAccel() function should actually be fabs(). Abs() returns an int, which would probably be 9, causing the if conditions to never be true. 
There are also 3 abs() calls in the calibrateMag() function, which I imagine should also be fabs().

Aerugo2272, 2019-10-07:
> @Aerugo2272 @nmk456 I found the problem. All the abs() calls in the calibrateAccel() function should actually be fabs(). Abs() returns an int, which would probably be 9, causing the if conditions to [...]

Biases still seem to be outputting 1 but the biases are indeed showing correct looking values now so that's a plus!

dobretotamje, 2020-07-16:
Found another problem in calibrateAccel, in the past there was copy paste error (line 756) - this was fixed in 2018, nevertheless the library downloaded from Arduino IDE labeled as version 1.0.1 still constains this issue.

ChisholmKyle, 2020-10-14:
The Arduino.h header actually undefs `abs` and creates it's own:

```c
// undefine stdlib's abs if encountered
#ifdef abs
#undef abs
#endif

#define min(a,b) ((a)<(b)?(a):(b))
#define max(a,b) ((a)>(b)?(a):(b))
#define abs(x) ((x)>0?(x):-(x))
```

Which I think is a real problem when people expect a standard library call from <math.h>. I think it was made to "help" new programmers unfamiliar with C standard implementation. Not to mention C++ `std::abs` can also handle floating point values.

ChisholmKyle, 2020-10-14:
Not sure if this helps, but bias values are being used before they're initialized:

```cpp
class MPU9250 {
    // ...
    float _axb, _ayb, _azb;
    float _axs = 1.0f;
    float _ays = 1.0f;
    float _azs = 1.0f;
    // ...
};
```

Then used on line 855
```cpp
_axbD += (getAccelX_mss()/_axs + _axb)/((double)_numSamples);
```

Ananfruit, 2020-10-26:
Any progress? Can you share the code when you calibrate your accel and gyro, please? Thanks in advance!

ChisholmKyle, 2020-10-26:
I made a pull request with a couple changes: https://github.com/bolderflight/MPU9250/pull/76
But I haven't tested it properly yet...

ChisholmKyle, 2020-10-26:
I wonder too the use of `double` does not actually mean double precision for Arduino avr, `double` is actually also 32-bit `float`.
So this code does not help with precision and accumulating a proper average at double precision:

```cpp
  // take samples and find bias
  _gxbD = 0;
  _gybD = 0;
  _gzbD = 0;
  for (size_t i=0; i < _numSamples; i++) {
    readSensor();
    _gxbD += (getGyroX_rads() + _gxb)/((double)_numSamples);
    _gybD += (getGyroY_rads() + _gyb)/((double)_numSamples);
    _gzbD += (getGyroZ_rads() + _gzb)/((double)_numSamples);
    delay(20);
  }
  _gxb = (float)_gxbD;
  _gyb = (float)_gybD;
  _gzb = (float)_gzbD;
```

flybrianfly, 2021-02-15:
Calibration functionality was removed from this library.

closed 2021-02-15 by flybrianfly