bolderflight/invensense-imu #57: Magnetometer calibration breaks if biases and scales were set before

dbuezas, 2019-08-20:
During calibration, the samples from the magnetometer are corrected using the scales and biases already stored.

e.g.: line 897:
```
_hxfilt = (_hxfilt * ((float)_coeff - 1) + (getMagX_uT() / _hxs + _hxb)) / ((float)_coeff);
```
This results in the new biases implicitly being relative to the old ones (which breaks them). A way to test it is to calibrate twice on a row, which results in different values.

I think the same line should be:
```
_hxfilt = (_hxfilt * ((float)_coeff - 1) + getMagX_uT()) / ((float)_coeff);
```
instead.

Amazing lib btw :)

korbinianbauer, 2020-07-07:
Any updates on this?

willmendil, 2020-12-31:
I've looked at the code, trying to port it to golang, i don't see what `_hxfilt`, `_hyfilt`, `_hzfilt` are as they are not declared to a given value if i'm not mistake.

flybrianfly, 2021-02-15:
Calibration functionality was removed from this library.

closed 2021-02-15 by flybrianfly