bolderflight/invensense-imu #58: wrong heading angle from the magnetometer

alieng1994, 2019-09-23:
Hi , thanks for the great job you doing my problem is I'm getting wrong value for heading angle and I,m using your library and calibration method to calculate the headeing so can you help me?
thanks

bevinhex, 2020-01-04:
first make sure you get the right axis, if you rotate the sensor on z axis, z axis magnetometer value should stay pretty much same, same goes for other axis, so that you can map the output of the library to right axis for your program. in my case Y & X is switched.
secondly, you need to find the right direction of each axis, I install and app on android for seeing magnetometer data, its x axis should be right side, so I move a magnet closer to right side of the phone, see the sensor output change on x axis, then I do the same for sensor's determined x axis, if it is increasing as phone did, it is correct, if it is doing opposite, just put minus to the library output.
In case you want to calibrate it afterwards, there is a program called "magneto1.2", it was for windows, but I'm able to use it with wine.
Hope this helps.

Ananfruit, 2020-10-26:
Can you share the code when you calibrate your accel and gyro, please? Thanks in advance!

alieng1994, 2020-10-26:
> first make sure you get the right axis, if you rotate the sensor on z axis, z axis magnetometer value should stay pretty much same, same goes for other axis, so that you can map the output of the [...]

Thank you for your help the was my sensor is fake one

alieng1994, 2020-10-26:
> Can you share the code when you calibrate your accel and gyro, please? Thanks in advance!

I used app on widows called magneto 1.2 to calibrate

closed 2021-02-15 by flybrianfly