bolderflight/invensense-imu #59: Does the library work on Arduino Nano ?

Elemyr, 2019-10-31:
After uploading the basic I2C sketch on an arduino Nano, I get no messages at all in the serial monitor. The sketch freezes whenever it gets to a part using this library. It uploads fine, with no warnings. However, using the same sketch on a Uno does work. Reinstalling the library didn't work.

Is this library supposed to be working on a Nano ? I've seen in other posts that it apparently is, but can't get it to work.

AzureSparkle, 2019-11-20:
I got it to work fine, however it's quite easy to use up too much memory on a nano which causes the system to become unstable

Elemyr, 2019-11-20:
I still don’t know exactly what was the reason but I had other sensors plugged to the Nano. Removing any made it work again, I guess I was drawing too much current or something. Anyway, it now works fine :)

closed 2019-11-20 by Elemyr