bolderflight/invensense-imu #60: sample rate higher than 1khz

yo90bosses, 2019-11-04:
Im unable to get a sample rate higher than 1khz. It seems locked to that because im able to get data at max exacly 1000 times a second. Any way to increse this limit? I should be able to get 8khz

flybrianfly, 2019-11-04:
Correct, this library only supports up to 1 kHz. You'd have to bypass the DLPF to reach a higher rate.

closed 2019-11-04 by flybrianfly

yo90bosses, 2019-11-14:
Is there any way i could modify this library to get 8khz? Do i only need to change what it writes to the registers or would that mean rewriting a lot of code?

flybrianfly, 2019-11-14:
I think you would need to remove or modify the DLPF setting in the begin method to get your desired rate. The appropriate bytes to send would be in the MPU-9250 datasheet.