bolderflight/invensense-imu #62: Values from FIFO not correct

yo90bosses, 2019-11-22:
Im trying to change my code to use the FIFO. The mpu9250 is connected to an esp32 over SPI and the example sketch works fine but if i enable the gyro and magnetometer some numbers are totally out of range and extremely noisy. 
Is there something im missing or is this an issue?

yo90bosses, 2019-11-26:
If I only enable the gyro and accel it works but I I use a seperate library for an HMC5883l over I2C then the numbers are incorrect. The gyro seems to be mixed with the accelerometer.

flybrianfly, 2019-11-26:
Regarding your first post, maybe the environment is very noisy for the magnetometer?

yo90bosses, 2019-11-26:
> Regarding your first post, maybe the environment is very noisy for the magnetometer?

Ne that can't be because if I'm not using the fifo it works fine. It's as if the incorrect bytes from the fifo are being used because the output from the gyro on two axises are being affected by acceleration and rotation. But in the x direction it's sometimes fine depending on what is enabled and is seems to even be effected by other libraries.
I'm guessing there is something wrong with changing the bytes to floats. 
Since the example sketch doesn't seem to work it looks like it's a general problem with the library or a compatibility issue with the esp32.

flybrianfly, 2019-11-26:
Works fine on the Teensy line, but has never been tested on ESP32.

yo90bosses, 2019-12-18:
> Works fine on the Teensy line, but has never been tested on ESP32.

Turned out to be general problem and was tested on the Teensy 3.6 and had the same issue.
https://forum.pjrc.com/threads/58688-Teensy-4-0-Clock-speed-influences-delay-and-SPI/page2
In post #35 is the results from testing on Teensy 3.6 and post #32 Contains the code that was tested.

mjs513, 2019-12-19:
Brian
Been playing with FIFO for the last day and a half and can not figure out the issue.  If I select Accel and temp it seems to be working fine, i.e, giving me correct accel values (-10m/s in z-axis, checked other rotations as well).  If I then try and select accel & gyro or accel/gyro/temp it starts giving very strange values.  Specifically accels now look like 0.3 or 0.4, if you add in mag the values jump up and go crazy like in the referenced post.  If you scroll down you will some other data from FIFO.

I also put in a bunch printf into the readFIfo function and other than the values being all messed up it all looks good.

It looks like the _buffer is getting messed up but can't figure out what the issue is.

mjs513, 2019-12-21:
Brian
Figured out the problem: see this post https://forum.pjrc.com/threads/58688-Teensy-4-0-Clock-speed-influences-delay-and-SPI?p=224387&viewfull=1#post224387.  But basically:
```
  //gather 50 samples at 980ms for accel
  //gather 50 samples at 980ms for accel/temp
  // gather 40 samples of data @800ms for accel/gyro
  // gather 35 samples of data @700ms for accel/gyro/temp
  // gather 25 samples of data @500ms for accel/gyro/mag
  // gather 23 samples of data @450ms for accel/gyro/mag/temp
```

flybrianfly, 2021-02-15:
Removed the FIFO functionality from the library.

closed 2021-02-15 by flybrianfly