bolderflight/invensense-imu #63: Support for Teensy 4.0

yo90bosses, 2019-12-11:
I just tried to get the library working with the new Teensy 4.0 board over SPI (SPI bus 0) with an MPU9250. I've been getting errors (-1 on .begin()) when running the example sketches. Has it been tested on the new Teensy board? Would be great if it worked because of the small form factor and very powerful processor.

yo90bosses, 2019-12-12:
Turned out that the ground connection to the mpu9250 was broken. But im now getting a -2 error. Is there a reason for this?

yo90bosses, 2019-12-12:
I found out that if i change the cpu frequency to a value between 24mhz to 396 the sketch is able to run. I opened the last issue about the libraries FIFO function not working with the ESP32 if i enable the anything other than the accelerometer. Im getting the exact same error with the new Teensy 4.0 board as i had with the ESP32. To me it looks like the library is having issues with the FIFO and maybe the SPI connection.

yo90bosses, 2019-12-18:
Turned out to be a problem with the Teensy 4.0. Bug was fixed

closed 2019-12-18 by yo90bosses

Bsschdi-tech, 2019-12-18:
Hey yo90bosses, i have the same Problem, I get the -2 error. How you fix your Problem?

yo90bosses, 2019-12-18:
> Hey yo90bosses, i have the same Problem, I get the -2 error. How you fix your Problem?

If youre using the spi bus then its because the teensy 4.0 currently has a problem with its spi bus and this library. Check out this thread where i posted the problem:
https://forum.pjrc.com/threads/58688-Teensy-4-0-Clock-speed-influences-delay-and-SPI

Currently is only works if you use the I2C bus or change the clock speed to 150mhz

Bsschdi-tech, 2019-12-20:
I change the Clockspeed to 150MHZ, but it still didn't Work.
This Part in the Libory is the right section or?
   const uint32_t SPI_LS_CLOCK = 150000000;  // 1 MHz
   const uint32_t SPI_HS_CLOCK = 150000000; // 15 MHz

yo90bosses, 2019-12-20:
> I change the Clockspeed to 150MHZ, but it still didn't Work.
> This Part in the Libory is the right section or?
> const uint32_t SPI_LS_CLOCK = 150000000; // 1 MHz
> const uint32_t SPI_HS_CLOCK = 150000000; // 15 MHz

The library needs to be modified to work reliably. Check in the pull requests where it was modified to work with teensy 4.0. A delay() was added to the spi read and write functions.