bolderflight/invensense-imu #66: Calibration Values not Working on STM32

W3AXL, 2020-01-07:
Trying to get a basic compass working on an STM32F103 using the stm32duino core.

Utilizing the code provided in #33 results in a nonworking heading reading (always 0). Excerpts of my adaption of the code (essentially verbatim with a few small tweaks) utilizing EEPROM reading & writing for storage of the values.

Loading of Mag Cal Values (runs at MCU startup)
```
  // Compass Calibration Reading
  uint8_t eeprom_buffer[24];
  for (unsigned int i = 0; i < sizeof(eeprom_buffer); i++ ) {
    eeprom_buffer[i] = EEPROM.read(i);
  }
  memcpy(&hxb, eeprom_buffer, sizeof(hxb));
  memcpy(&hyb, eeprom_buffer + 4, sizeof(hyb));
  memcpy(&hzb, eeprom_buffer + 8, sizeof(hzb));
  memcpy(&hxs, eeprom_buffer + 12, sizeof(hxs));
  memcpy(&hys, eeprom_buffer + 16, sizeof(hys));
  memcpy(&hzs, eeprom_buffer + 20, sizeof(hzs));
  IMU.setMagCalX(hxb, hxs);
  IMU.setMagCalY(hyb, hys);
  IMU.setMagCalZ(hzb, hzs);
```
Calibration Method & saving of values to EEPROM (run by user whenever desired)
```
// Compass Calibration Method
void calibrateCompass() {
  int stat = IMU.calibrateMag();
  if (stat > 0) {
    // get cal values and save to a buffer
    uint8_t eeprom_buffer[24];
    float value;
    // bias values
    value = IMU.getMagBiasX_uT();
    memcpy(eeprom_buffer, &value, sizeof(value));
    value = IMU.getMagBiasY_uT();
    memcpy(eeprom_buffer + 4, &value, sizeof(value));
    value = IMU.getMagBiasZ_uT();
    memcpy(eeprom_buffer + 8, &value, sizeof(value));
    value = IMU.getMagScaleFactorX();
    memcpy(eeprom_buffer + 12, &value, sizeof(value));
    value = IMU.getMagScaleFactorY();
    memcpy(eeprom_buffer + 16, &value, sizeof(value));
    value = IMU.getMagScaleFactorZ();
    memcpy(eeprom_buffer + 20, &value, sizeof(value));
    for (unsigned int i = 0; i < sizeof(eeprom_buffer); i++) {
      EEPROM.write(i, eeprom_buffer[i]);
    }
    // Set compass cal values
    IMU.setMagCalX(hxb,hxs);
    IMU.setMagCalY(hyb,hys);
    IMU.setMagCalZ(hzb,hzs);
    // Finalize
    cmpStr = "Success!";
  } else if (stat < 0) {
    cmpStr = "Cal failed!";
  }
  cmpStat = 0;
}
```
If the initial loading of cal values is commented out, the compass works just fine (but obviously uncalibrated). If a user calibration is run (the second function posted) the compass once again breaks and gets stuck reading 0. These two points tell me that a) my other code to process and calculate the heading is OK, and b) something about setting the calibration factors is breaking other things with the MPU.

Happy to provide more info if needed.

As a bonus, here's the code, again from #33, for calculating the compass heading.
```
    // Get raw values
    float ax = IMU.getAccelX_mss();
    float ay = IMU.getAccelY_mss();
    float az = IMU.getAccelZ_mss();
    float hx = IMU.getMagX_uT();
    float hy = IMU.getMagY_uT();
    float hz = IMU.getMagZ_uT();
    // Normalization
    float h = sqrtf(hx*hx + hy*hy + hz*hz);
    float a = sqrtf(ax*ax + ay*ay + az*az);
    hx /= h;
    hy /= h;
    hz /= h;
    ax /= a;
    ay /= a;
    az /= a;
    // Euler angles
    float pitch_rad = asinf(ax);
    float roll_rad = asinf(-ay / cosf(pitch_rad));
    float yaw_rad = atan2f(hz * sinf(roll_rad) - hy * cosf(roll_rad), hx * cosf(pitch_rad) + hy * sinf(pitch_rad) * sinf(roll_rad) + hz * sinf(pitch_rad) * cosf(roll_rad));
    float hdg_rad = constrainAngle360(yaw_rad + (hdgCorr * PI / 180));
    int hdg = hdg_rad * 180 / PI;
    avgHdg = rollingAverage(avgHdg, hdg);
    // Graphics Update
    CMP_update(int(avgHdg)+hdgCorr);
```

W3AXL, 2020-01-07:
Additional testing: if I comment out the setting of cal values after the user calibration, and run a calibration (moving in a figure 8 pattern) I get further strange behavior of the compass. Heading readings are no longer even close to correct and jump around all over the place, and also are dramatically affected by tilt whereas before the readings worked independent of sensor orientation (which is the expected behavior using the calculation code above)

flybrianfly, 2021-02-15:
Calibration functionality was removed from this library.

closed 2021-02-15 by flybrianfly