bolderflight/invensense-imu #68: MPU9265 Gyro always Zero.

lkaupp, 2020-06-03:
Hi Folks,

got the new MPU9265, i2c_scanner device is there, .begin() .calibration() of all Gyro, Accel, Mag, is working, values look good, besides the Gyro-values:

`

0
--
0,01
0,03
0
0,01
0,03
0
0,02
0,03
-0,01
0,02
-0,11
-0,01
-0,03
0,01
0,02
-0,02
0,02
-0,02

`
The values are always around zero. Anything that i have forgotten, set i need to set manually to get gyro-sensor working?

engrnoor, 2020-11-19:
I am having similar issue with MPU9250
All gyros have 0 readings

with Arduino nano in I2C configuration. 
I used the Basic_I2C Example

The output of the serial monitor is shown bellow

[image]

hummyann, 2020-12-10:
I'm got this issue too. It seems not an issue on this library - I just tried to read the Gyro register without this library and i got all 0 on gyros. If anyone could help? Or i had got a crappy sensor?

lkaupp, 2020-12-11:
Two observations i could share. Gyroscope activity is always around +-0.XX. The gyro activity (attached to a robot arm)  is synchronous with other movement values of the sensor and therefore i think its legit. Furthermore, I think the given value is not that helpful and you can compute the angle with the +-0.xx value (idk the suitable formula yet, have to search for some code in public github repos). I had another mistake in initialization of the sensors in the beginning (i got chaotic sensor values sometimes [way above and down specification]). Make sure you generate an object per physical sensor and calibrate each object with its physical counter part. At the beginning i though i would be smart and reuse the objects but the calibration process is unique and every physical sensor gets somehow different init values from the sensor, which in contrary makes the obj unique and not reuseable among other sensors of the same type and same manufacturer!

mustafakisacik, 2021-01-11:
In mpu9250, gyroscope x, y, z axes give -9223372036854775808.08, -9223372036854775808.08, -9223372036854775808.08  value at Kalman filter output. Why am I getting these values. (Baud rate value 230400, gyro scales 250 dps )

ddv2005, 2021-01-23:
> I am having similar issue with MPU9250
> All gyros have 0 readings

Mine the same - all gyros always zero. And looks like it is sensor issue. I bought mine from HiLetgo at Amazon and all 3 sensors has some issues: 1 sensor just not detected, 2 sensors all Gyro values are zero. Also I bought another 2 MPU9250 from another seller and both works fine. I have no idea why all sensors from HiLetgo are bad

mustafakisacik, 2021-01-23:
I'll buy a different one and try it. I hope I don't get errors again.

flybrianfly, 2021-02-15:
The gyros measure rotational rate, not orientation, so they should be very small if the sensor is stationary since you are basically just seeing sensor noise. If they are perfectly zero, it's likely a bad sensor.

closed 2021-02-15 by flybrianfly