bolderflight/invensense-imu #77: Initialisation error

MartinBzm, 2020-10-16:
Hi,
When trying the example Basic_I2C the compiler throws an error at:

`// an MPU9250 object with the MPU-9250 sensor on I2C bus 0 with address 0x68
MPU9250 IMU(Wire,0x69);`

`Basic_I2C:28:22: error: no matching function for call to 'MPU9250_<TwoWire>::MPU9250_(TwoWire&, int)'
 MPU9250 IMU(Wire,0x69);`

the `#include <Wire.h>` is also present.

Is something missing? Am I missing something?
Thanks in advance!
Martin

flybrianfly, 2021-02-15:
That's weird, can you provide more information about the microcontroller you are using?

MartinBzm, 2021-02-15:
Hi. It has been a while so I may not have the details right. I was very busy with Arduino Mega and Micro boards then and I think it was on the Mega 2560. I'll try and find some time soon to re-create the setup and try it again, I'm sure I've saved it somewhere ;-)

closed 2021-02-28 by flybrianfly

wollewald, 2021-03-06:
@MartinBzm @flybrianfly I had the same issue and was very surprised, because initially it worked. I found the reason : I have tried several MPU9250 libraries and there was another one using the file name "MPU9250.h".  I would be better to choose a more individual name. 
Thanks for the library!