bolderflight/invensense-imu #81: MPU9250 Mag always zero with WakeupOnMotion enabled

erberj, 2020-12-03:
Hi,
wenn enabling 
 IMU.enableWakeOnMotion(40.0,MPU9250::LP_ACCEL_ODR_7_81HZ);

 delay(1000);
  IMU.readSensor();
  delay(1000);
  imx = IMU.getMagX_uT();
  imy = IMU.getMagY_uT();
  imz = IMU.getMagZ_uT();

returns zero for all values. Even before the MCU MCU Atmel 8-bit AVR: ATMega328P-AU (32Kb Flash/2Kb RAM) goes into sleep the first time. What coud go wrong? If I define an interrupt without the wake part, it works fine.
The MPU is connected via I2C

Best Regards

zcsrf, 2020-12-29:
Hi,

When you set the enableWakeOnMotion, the Gyro and Magnetic sensor are disabled.
You can check the function for better understanding of what is does.

Only do the enable of WoM before going to sleep.

flybrianfly, 2021-02-14:
WoM is no longer supported.

closed 2021-02-14 by flybrianfly