bolderflight/invensense-imu #89: Same Board?

RedSoxFan04, 2021-04-06:
The board this is for, is it the same as this one? https://www.amazon.com/HiLetgo-Gyroscope-Acceleration-Accelerator-Magnetometer/dp/B01I1J0Z7Y/ref=sr_1_1_sspa?dchild=1&keywords=InvenSense+MPU-9250&qid=1617741587&sr=8-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExN0FCVzdSNklDVUVaJmVuY3J5cHRlZElkPUEwNzI4MTE4NExFVEFURzdYODNHJmVuY3J5cHRlZEFkSWQ9QTAxMDY3NDUzT01PT0ZYS0pUVDNTJndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==

flybrianfly, 2021-04-06:
It's for the MPU-9250 sensor, not any particular breakout board. I would use caution and prefer to purchase a breakout board from a reputable vendor, like Sparkfun, there are a lot of fakes on the market.

closed 2021-04-06 by flybrianfly

RedSoxFan04, 2021-04-07:
> It's for the MPU-9250 sensor, not any particular breakout board. I would use caution and prefer to purchase a breakout board from a reputable vendor, like Sparkfun, there are a lot of fakes on the market.

I don't really have any experience in this area. I'm just curious about these boards because I'm looking for a way to change the orientation of my monitor in software when I physically rotate it. Since I don't have any experience with this sort of thing, I don't really have any idea of reputable brands - aside from Sparkfun, which you mentioned.

flybrianfly, 2021-04-07:
Besides Sparkfun and Emsensor, I would also trust Pesky Products:
https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-mpu9250/

If Adafruit carried it, they would be on the trusted list as well.