bolderflight/invensense-imu #96: Reading erros

TahaMoncef, 2021-12-10:
Hello sir, thank you for your help, however, unfortunately when I try to test the I2C code given in the example, I get the following readings which I believe are wrong:
[image]

flybrianfly, 2021-12-10:
Can you post a picture of your sensor and setup?

TahaMoncef, 2021-12-10:
> Can you post a picture of your sensor and setup?

the Vcc pin is connected to the 5v of the arduino
the ground pin is connected to the ground of the 5v supply
SCL pin connected to A21 since i'm using Arduino mega 2560
SDA pin to A20
[image]

TahaMoncef, 2021-12-10:
> Can you post a picture of your sensor and setup?

[image]

TahaMoncef, 2021-12-10:
the Vcc pin is connected to the 5v of the arduino
the ground pin is connected to the ground of the 5v supply
SCL pin connected to A21 since i'm using Arduino mega 2560
SDA pin to A20
[image]
[image]

flybrianfly, 2021-12-10:
Seems weird, the total acceleration is higher than it should be (i.e. around 12 m/s/s instead of 9.8 m/s/s). Also seems weird that you're not getting magnetometer data as often as I would expect. I would probably touch up the soldering, install the latest version of this library, and test while ensuring that the sensor is resting on a table.

TahaMoncef, 2021-12-11:
I thought about a potential solution: Taking few seconds when the code starts running to calculate the mean value of each sensor(on each axe) and then consider these "means" as an offset.
Then, I can simply compensate for the shift in the raw values by adding the missing values due to the offset.
However, I'm not so sure of this solution since I'm not aware if this offset is linear or not.
PS: i am using the latest version of the library.

flybrianfly, 2021-12-11:
It's not the newest version of the library, we use a namespace and the class object isn't fully capitalized, which is different than the screenshot posted. I don't think it's the code, but helps to get on the same baseline for any troubleshooting.

You should absolutely calibrate it anyway, but my expectation would be that the value is closer to {0, 0, -9.8} at rest with the sensor z axis aligned with gravity. You could calibrate it by moving the sensor very slowly in a spherical motion and then doing a least squares estimate of the resulting data to get a bias and scale factor for each axis (exciting each axis with +g and -g enables you to get both a bias and a scale factor).

TahaMoncef, 2021-12-11:
Indeed, i was just trying to test multiple libraries, unfortunately i keep getting the same error even with this latest version.
One more thing, i've noticed another measurement error: the x-axis column and y-axis column measurements are swapped which i believe that i'm the only one getting this weird results.

TahaMoncef, 2021-12-17:
hello sir, so I bought a new MPU chip, and tried to do the soldering very carefully, this time it worked correctly.
One last thing, do you have any sort of documentation or guide to calibrate the magnetometer, I'm trying to build up a compass, and all that is left is to calibrate the magnetometer?
Thank you for your help.

flybrianfly, 2021-12-17:
You'll want to slowly move the magnetometer so that each axis traces a sphere while recording data (this ends up being kind of a figure 8 shape that you may be familiar with from calibrating the magnetometer for phone apps or a smart watch). Then use a least squares method to find a scale factor and bias for each axis to fit the data to a sphere. That's it! You'll need to re-calibrate any time the electromagnetic environment changes.

There's also a bunch of web pages with instructions and video demos if you search for "how to calibrate a magnetometer".

closed 2021-12-20 by flybrianfly