bolderflight/invensense-imu #100: Eigen.h file is missing

frozenmafia, 2022-02-25:
I am trying to use this library with esp32 but compiler shows eigen.h is missing.

flybrianfly, 2022-02-25:
Can you post the full outputs from Arduino? I don't have an ESP32 to test against.

frozenmafia, 2022-02-26:
[image]
This is not compiling. Problem is that Eigen.h is missing.
[Here in mpu9250.h eigen.h is included](https://github.com/bolderflight/mpu9250/blob/01d19b8c086243170125a2d2dfb437353d30bd28/src/mpu9250.h#L38)

But there is [no file in named eigen.h ](https://github.com/bolderflight/mpu9250/tree/main/src).
I have just switched from atmega328p to esp32 and this library is my last hope 
Thanks for replying

frozenmafia, 2022-02-27:
Thanks it worked! Next time I will read instructions more carefully.

closed 2022-02-27 by frozenmafia