bolderflight/invensense-imu #102: Does setSrd() only work with interrupt?

ErfanLab, 2022-03-06:
Hi, I have a problem with readSensor() function, I thought this would work with setSrd() and whenever there is a data (according to the sample rate) it returns positive. But with testing it I saw that this is not true.
What is the correct way of using setSrd()? Is it meant to be used with interrupt? 
By the way, very nice and useful library. Thanks a lot.

flybrianfly, 2022-03-06:
Can you post more information:

- What version of the library are you using?
- What microcontroller?
- What breakoutboard? Are you using SPI or I2C?
- Post the simplest example code that demonstrates the problem.

flybrianfly, 2022-03-15:
I'm not sure what version fixed it, but the SRD works as expected in the current version of the library, v5.6.0

closed 2022-03-15 by flybrianfly