bolderflight/invensense-imu #105: Glitches on Gyro data

mmcgraw74, 2022-03-22:
I ran your i2c example, but I added Serial print statements for Arduino Serial Plotter labels, adding offsets so the traces don't overlap.

if (imu.Read()) {

            Serial.print("Max:250,");    
            Serial.print("NewIMU:");    
            Serial.print(imu.new_imu_data()*150);
            Serial.print(",");
            
            Serial.print("NewMAG:");    
            Serial.print(imu.new_mag_data()*100);
            Serial.print(",");

            Serial.print("AccX:");    
            Serial.print(imu.accel_x_mps2());
            Serial.print(",");
            
            Serial.print("AccY:");    
            Serial.print(imu.accel_y_mps2()-20);
            Serial.print(",");
            
            Serial.print("AccZ:");    
            Serial.print(imu.accel_z_mps2()-40);
            Serial.print(",");
            
            Serial.print("GyrX:");    
            Serial.print(imu.gyro_x_radps()-20);
            Serial.print(",");
            
            Serial.print("GyrY:");    
            Serial.print(imu.gyro_y_radps()-40);
            Serial.print(",");
            
            Serial.print("GyrZ:");    
            Serial.print(imu.gyro_z_radps()-60);
            Serial.print(",");
            
            Serial.print("MagX:");    
            Serial.print(imu.mag_x_ut());
            Serial.print(",");
            
            Serial.print("MagY:");    
            Serial.print(imu.mag_y_ut()-20);
            Serial.print(",");
            
            Serial.print("MagZ:");    
            Serial.print(imu.mag_z_ut());
            Serial.println("Min:-250");    
                        
            //Serial.print(imu.die_temp_c());
            //Serial.print("\n");

when I run this example with my ESP32 Dev Board and MPU9250 board, I use Arduino Serial Plotter and get the following plot:

[image]

Searching the web - I found this post about gyro glitches:
https://forums.adafruit.com/viewtopic.php?f=19&p=531590

I wonder if the issue is similar - the code is getting interrupted between individual MPU register reads, and should read whole blocks of MPU i2c registers to prevent this kind of issue.

mmcgraw74, 2022-03-22:
The glitches appear on different sensors at random intervals.  I had tried filtering algorithms on the data but the glitches are large magnitude.

mmcgraw74, 2022-03-22:
The green spikes are the plot of the boolean imu.new_mag_data(), which looks correct - and is not the issue I'm describing.

The issue in the plotter output above is on the imu.mag_y_ut() output.

mmcgraw74, 2022-03-22:
Here is one more snapshot - I commented out the New Mag signal.
You can see a glitch on MagX followed later by a glitch on MagY.
[image]

mmcgraw74, 2022-03-22:
Sorry, same plot as before.
Here is the other plot:
[image]

flybrianfly, 2022-03-22:
Can you post the full code that you are running and confirm that you are running the newest version of this library (i.e. v5.6.0)? We burst read the IMU, but we have had issues in the past with the ESP32 not working with the I2C repeated start. I don't have an ESP32 board to test, but will try to see if I can replicate the issues on the Teensy boards that I have. Is there any potential to try reading the sensor with SPI? That might help us determine if it's an issue with the I2C communication or something else.

mmcgraw74, 2022-03-23:
I confirm I'm running Blolder Flight Systems MPU9250 v 5.6.0 Arduino library.

mmcgraw74, 2022-03-27:
I rewired my board to use ESP32 VSPI with the MPU-9250 board instead of i2c.

I ran your SPI example and didn't see any glitches (left running for 45 minutes.  Here is the serial plot:

[image]

I also tried the fifo_SPI example and it looks like LOTS of spikes:

[image]