bolderflight/invensense-imu #111: Why inverted axis for gyro and accelerometer

tcerqueira, 2022-06-07:
Months using this library with some problems along the way and I just realized for each gyro axis the accelerometer axis is inverted. I have been assuming the same frame of reference for both and it explains a lot of my problems. Is this intended behaviour or am I missing something?

flybrianfly, 2022-06-08:
The accel, gyro, and mag all use the [coordinate frame depicted](https://github.com/bolderflight/mpu9250#sensor-orientation). If I accelerate the gyro in the positive X direction, I get a positive acceleration. Positive gyro is likewise a right-hand rotation around the axis, so I'm not seeing where the inversion is coming from.

flybrianfly, 2022-06-09:
I think there's a misunderstanding; an accelerometer at rest will measure 9.80665 m/s/s of upwards acceleration. There is a [Wikipedia article](https://en.wikipedia.org/wiki/Accelerometer) that gives a decent explanation. You can try accelerating the IMU by hand in different directions (i.e. accelerating forward along the X axis) and I think you can see that the accelerometer and gyro axis are in agreement.

tcerqueira, 2022-06-09:
Oh I thought the accelerometer data was an acceleration vector so I was expecting a positive value for gravity since the z-axis is down. Wrong assumption I guess, thanks for clarifying.

flybrianfly, 2022-06-09:
It is an acceleration vector, except for gravity.

Founder CEO, Bolder Flight Systems
http://bolderflight.com
Ph: (719) 246 - 4559

closed 2022-06-10 by flybrianfly