bolderflight/invensense-imu #115: Incompatible with AVR architecture ?

sterretjeToo, 2022-10-07:
The following error message shows when compiling the i2c or drdy_spy examples when compiling for the Arduino Mega.
```
In file included from C:\Users\sterretje\AppData\Local\Temp\.arduinoIDE-unsaved202297-20044-3fuz8n.md4do\i2c\i2c.ino:26:0:
c:\Users\sterretje\Documents\Arduino\libraries\Bolder_Flight_Systems_MPU9250\src/mpu9250.h:36:10: fatal error: cstddef: No such file or directory
 #include <cstddef>
          ^~~~~~~~~
compilation terminated.
```
The above was error occured when using IDE2.0; I've also tried using portable installs of 1.8.19 and 1.6.6.

Trying to compile was triggered by the following topic on forum.arduino.cc https://forum.arduino.cc/t/i-am-trying-to-connect-mpu9250-with-my-arduino-uno-and-i-am-getting-this-error-while-uploading-the-code/982106

According to https://www.arduino.cc/reference/en/libraries/bolder-flight-systems-mpu9250/, the library is compatible with ALL architectures. But *cstddef* never seems to have been part of the AVR architecture according to https://github.com/arduino/Arduino/issues/5209

Regards, sterretje

jetsup, 2022-10-07:
I have encountered the same problem. I used the ```locate``` command and the ```cstddef``` library is in my system, but I still can't figure out why it can't be loaded.

flybrianfly, 2022-10-07:
@sterretjeToo @jetsup, v6 and newer of this library supports AVR and should work out of the box. Please download directly from GitHub, the Arduino Library manager does a poor job of grabbing library updates. Let me know if this doesn't work, we specifically developed and tested that version against AVR.

@jetsup, it doesn't matter if cstddef is on your system, the AVR compiler cannot use it.

jetsup, 2022-10-11:
V6.0.2 worked for me... Okay @flybrianfly, I thought it did depend on it. Thank you.

closed 2023-03-17 by flybrianfly

per1234, 2024-10-23:
> the Arduino Library manager does a poor job of grabbing library updates.

@flybrianfly the poor job is not from Library Manager, but from you. You changed the library name from "Bolder Flight Systems MPU9250" to "Bolder Flight Systems InvenSense IMU":

https://github.com/bolderflight/invensense-imu/commit/8f049251063684705cbee1a73dd64c365a138576#:ro5:

```diff
--- a/library.properties
+++ b/library.properties
- name=Bolder Flight Systems MPU9250
+ name=Bolder Flight Systems InvenSense IMU
```

The library name is its identifier in the Library Manager system so the library is locked to the name it had at the time of registration. Any release made under a different name is rejected by the Library Manager system due to being invalid.

So if you want your future releases to be indexed, you must either change the value of the `name` property of the `library.properties` metadata file back to "Bolder Flight Systems MPU9250", or else submit a name change request:

https://github.com/arduino/library-registry/blob/main/FAQ.md#how-can-i-change-a-librarys-name