bolderflight/invensense-imu #116: Compiling IDE 2.0 and arduino nano 33 iot

Luporosso76, 2023-02-26:
HI,
when I compile I have the following error.
Can you help me understand the problem?
Thank you
Alessandro

#######################
In file included from /tmp/arduino-sketch-96B8BA06D0DBE4A712B21060E2BF6926/sketch/fifo_spi.ino.cpp:1:0:
/home/lupo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/complex:69:30: error: expected unqualified-id before 'const'
   template<typename _Tp> _Tp abs(const complex<_Tp>&);
                              ^
/home/lupo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/complex:69:30: error: expected ')' before 'const'
/home/lupo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/complex:69:30: error: expected ')' before 'const'
/home/lupo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/complex:601:5: error: expected unqualified-id before 'const'
     abs(const complex<_Tp>& __z) { return __complex_abs(__z); }
     ^
/home/lupo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/complex:601:5: error: expected ')' before 'const'
/home/lupo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/complex:601:5: error: expected ')' before 'const'
/home/lupo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/complex: In static member function 'static _Tp std::_Norm_helper<true>::_S_do_it(const std::complex<_Tp>&)':
/home/lupo/.arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/Arduino.h:66:16: error: expected unqualified-id before '(' token
 #define abs(x) ((x)>0?(x):-(x))
                ^
In file included from /home/lupo/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/Core:50:0,
                 from /home/lupo/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/eigen.h:36,
                 from /home/lupo/Arduino/libraries/Bolder_Flight_Systems_MPU9250/src/mpu9250.h:38,
                 from /tmp/.arduinoIDE-unsaved2023126-411621-4ib43m.yuken/fifo_spi/fifo_spi.ino:26:
/home/lupo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/complex: In function 'std::complex<_Tp> std::__complex_log(const std::complex<_Tp>&)':
/home/lupo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/complex:773:26: error: expected primary-expression before '(' token
     { return complex<_Tp>(log(std::abs(__z)), std::arg(__z)); }
                          ^
In file included from /tmp/arduino-sketch-96B8BA06D0DBE4A712B21060E2BF6926/sketch/fifo_spi.ino.cpp:1:0:
/home/lupo/.arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/Arduino.h:66:16: error: expected unqualified-id before '(' token
 #define abs(x) ((x)>0?(x):-(x))
                ^
/home/lupo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/complex: In function 'std::complex<_Tp> std::__complex_sqrt(const std::complex<_Tp>&)':
/home/lupo/.arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/Arduino.h:66:16: error: expected unqualified-id before '(' token
 #define abs(x) ((x)>0?(x):-(x))
                ^
/home/lupo/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/complex: In function '_Tp std::fabs(const std::complex<_Tp>&)':
/home/lupo/.arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/Arduino.h:66:16: error: expected unqualified-id before '(' token
 #define abs(x) ((x)>0?(x):-(x))
                ^
/home/lupo/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/MathFunctions.h: At global scope:
/home/lupo/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/MathFunctions.h:1509:5: error: expected unqualified-id before 'const'
 abs(const T &x) {
     ^
/home/lupo/.arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/Arduino.h:66:18: note: in definition of macro 'abs'
 #define abs(x) ((x)>0?(x):-(x))
                  ^
/home/lupo/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/MathFunctions.h:1509:5: error: expected ')' before 'const'
 abs(const T &x) {
     ^
/home/lupo/.arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/Arduino.h:66:18: note: in definition of macro 'abs'
 #define abs(x) ((x)>0?(x):-(x))
                  ^
/home/lupo/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/MathFunctions.h:1509:5: error: expected ')' before 'const'
 abs(const T &x) {
     ^
/home/lupo/.arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/Arduino.h:66:18: note: in definition of macro 'abs'
 #define abs(x) ((x)>0?(x):-(x))
                  ^
/home/lupo/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/MathFunctions.h:1517:5: error: expected unqualified-id before 'const'
 abs(const T &x) {
     ^
/home/lupo/.arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/Arduino.h:66:18: note: in definition of macro 'abs'
 #define abs(x) ((x)>0?(x):-(x))
                  ^
/home/lupo/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Core/MathFunctions.h:1517:5: error: expected ')' before 'const'
 abs(const T &x) {
     ^
[... 287 lines cut ...]
                   ^~~
/home/lupo/.arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/Arduino.h:66:16: error: expected unqualified-id before '(' token
 #define abs(x) ((x)>0?(x):-(x))
                ^
/home/lupo/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Eigenvalues/SelfAdjointEigenSolver.h:526:57: note: in expansion of macro 'abs'
         if (scaled_subdiag * scaled_subdiag <= (numext::abs(diag[i])+numext::abs(diag[i+1]))) {
                                                         ^~~
/home/lupo/.arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/Arduino.h:66:16: error: expected unqualified-id before '(' token
 #define abs(x) ((x)>0?(x):-(x))
                ^
/home/lupo/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Eigenvalues/SelfAdjointEigenSolver.h:526:78: note: in expansion of macro 'abs'
         if (scaled_subdiag * scaled_subdiag <= (numext::abs(diag[i])+numext::abs(diag[i+1]))) {
                                                                              ^~~
/home/lupo/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Eigenvalues/SelfAdjointEigenSolver.h: In function 'void Eigen::internal::tridiagonal_qr_step(RealScalar*, RealScalar*, Index, Index, Scalar*, Index)':
/home/lupo/.arduino15/packages/arduino/hardware/samd/1.8.13/cores/arduino/Arduino.h:66:16: error: expected unqualified-id before '(' token
 #define abs(x) ((x)>0?(x):-(x))
                ^
/home/lupo/Arduino/libraries/Bolder_Flight_Systems_Eigen/src/Eigen/src/Eigenvalues/SelfAdjointEigenSolver.h:851:19: note: in expansion of macro 'abs'
     mu -= numext::abs(e);
                   ^~~

exit status 1

Compilation error: exit status 1
##################

flybrianfly, 2023-02-26:
It compiles fine for me.
[image]

I would check that you're using the latest version of the libraries directly from GitHub, sometimes the Arduino IDE doesn't pickup version updates.

Luporosso76, 2023-02-26:
HI,
You are right. I installed the package from github and with that I can compile.
Arduino IDE loaded version 5.6.0

A thousand thanks
Alessandro

closed 2023-02-26 by Luporosso76