bolderflight/invensense-imu #119: status -5

Narek-939, 2023-07-08:
Hi, please help me, I get the error with status -5, and if my IMU is not MPU9250, how I can resolve this issue

flybrianfly, 2023-07-08:
1. You're using a really old version of the library, please download the latest directly from GitHub.
2. Only the MPU-9250, MPU-92500, and MPU-6500 are currently supported.

Narek-939, 2023-07-09:
I have changed my version and encountered to this error

Arduino: 1.8.19 (Linux), Board: "Arduino Nano, ATmega328P (Old Bootloader)"

MPU9250:42:1: error: 'Mpu9250' does not name a type
 Mpu9250 IMU(Wire,0x68);
 ^~~~~~~
/home/vis/Downloads/MPU9250/MPU9250.ino: In function 'void setup()':
MPU9250:51:12: error: 'IMU' was not declared in this scope
   status = IMU.begin();
            ^~~
/home/vis/Downloads/MPU9250/MPU9250.ino: In function 'void loop()':
MPU9250:63:3: error: 'IMU' was not declared in this scope
   IMU.readSensor();
   ^~~
exit status 1
'Mpu9250' does not name a type

So, I opened the files of this library and found that the class of MPU9250 is named "Mpu9250", I changed the name in my code but it gave me the same error.

Could you say me, what need to be the class name of that object  "Mpu9250 IMU(Wire,0x68);"

flybrianfly, 2023-07-09:
Which InvenSense IMU are you using? This library only works with the MPU-9250, MPU-9255, and MPU-6500.

Narek-939, 2023-07-10:
I am using the MPU9250