bolderflight/invensense-imu #120: Can't initialize I2C bus

brightproject, 2023-07-09:
Microcontroller `stm32f401ccu6`
Connecting `SDA-PB7`, `SCL-PB6`
IMU `MPU-9250`

```
#include "mpu9250.h"

/* Mpu9250 object, I2C bus,  0x68 address */
bfs::Mpu9250 imu(&Wire, bfs::Mpu9250::I2C_ADDR_PRIM);

void wakeup() {
  Serial.println("Motion");
}

void setup() {
  /* Serial to display data */
  Serial.begin(115200);
  while(!Serial) {}
  /* Start the I2C bus */
  Wire.begin();
  Wire.setClock(400000);
  /* Initialize and configure IMU */
  if (!imu.Begin()) {
    Serial.println("Error initializing communication with IMU");
    while(1) {}
  }
  /* 
  * Enable wake on motion with a threshold of 40 mg and an accel data rate of
  * 15.63 Hz 
  */
  imu.EnableWom(40, bfs::Mpu9250::WOM_RATE_15_63HZ);
  /* Attach the interrupt to pin 9 */
  pinMode(9, INPUT);
  attachInterrupt(9, wakeup, RISING);
}

void loop() {}
```
I get the following in the port monitor:
```
PS C:\Users\Admin> arduino-cli monitor -p COM6 -c baudrate=115200
Monitor port settings:
baudrate=115200
Connected to COM6! Press CTRL-C to exit.
Error initializing communication with IMU
```

flybrianfly, 2023-07-09:
Have you tried running an I2C scanner program on the microcontroller to verify that it can see the MPU-9250 on the I2C bus and check the I2C address?

brightproject, 2023-07-09:
> Have you tried running an I2C scanner program

Answered a[bove](https://github.com/bolderflight/invensense-imu/issues/120#issue-1795370881)

brightproject, 2023-07-09:
Why in the example, the bus config I`2C `is done like this
`imu.Config(&Wire, bfs::Mpu9250::I2C_ADDR_PRIM);`
And in the library in the class `mpu9250.h` like this
`Mpu9250(TwoWire *i2c, const I2cAddr addr`
...&Wire...TwoWire

flybrianfly, 2023-07-09:
TwoWire is the object type that Arduino defines for I2C. Somewhere in the core software for the board, there is some declaration like:
TwoWire Wire(...);
TwoWire Wire1(...);

For the library, we're just grabbing a pointer to the I2C bus in use, especially since many boards have multiple I2C buses.

I would also check the I2C address using the scanner to verify whether you should be using I2C_ADDR_PRIM or I2C_ADDR_SEC.

brightproject, 2023-07-09:
> Я бы также проверил адрес I2C с помощью сканера, чтобы убедиться, что вы должны использовать I2C_ADDR_PRIM или I2C_ADDR_SEC.
[image]
I am using this [core](https://github.com/stm32duino/Arduino_Core_STM32) for adruino IDE.
It has a `wire.h` library, and I had to remove the folder from the ARDUINO folder so that there is no conflict, using two libraries.
Later, I returned the folder with the `wire.h` library to the ARDUINO folder, and prescribed the use of the library
#include <Wire.h>
and
#include "Wire.h>"
But the result is the same.
It seems to me that the STM32 compiler does not understand instructions?
`static_cast`

brightproject, 2023-07-09:
If commented section `while` in code
```
  if (!imu.Begin()) {
    Serial.println("Error initializing communication with IMU");
    //while (1) {}
  }
  /* Set the sample rate divider */
  if (!imu.ConfigSrd(19)) {
    Serial.println("Error configured SRD");
    //while (1) {}
  }
```
I see a partial execution of the library code
```
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    32.58
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    32.39
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    32.39
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    32.63
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    32.63
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    32.63
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    32.58
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    32.58
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    32.73
```
[image]
Unit, apparently `Serial.print(imu.new_imu_data());`
And the numbers changing at the end, this is apparently `Serial.print(imu.die_temp_c());`
If disconnect MPU-9250 from SDA/SCL.
Then the output stops and lines appear.
```
Error initializing communication with IMU
Error configured SRD
```
I don't understand what is happening and why the [library](https://github.com/bolderflight/invensense-imu) cannot configure i2C.

brightproject, 2023-07-09:
The only library that we managed to launch for STM32 and arduino IDE.
I2C did not even specify contacts, the library itself determined them.
[image]
```
Accel: -0.01, 0.09, 1.01 g
Gyro: -4.57, 4.51, 1.65 dps
Mag: 0.00, 0.00, 0.00 uT
Time: 118769 ms

Accel: -0.02, 0.12, 1.01 g
Gyro: -4.70, 2.99, 2.01 dps
Mag: 0.00, 0.00, 0.00 uT
Time: 120776 ms

Accel: -0.02, 0.14, 1.01 g
Gyro: -4.45, 3.54, 2.01 dps
Mag: 0.00, 0.00, 0.00 uT
Time: 122785 ms

Accel: -0.04, 0.17, 1.00 g
Gyro: -4.76, 2.99, 2.01 dps
Mag: 0.00, 0.00, 0.00 uT
Time: 124796 ms
```
Questions only to your library - apparently it is not optimized for `stm32f401ccu6`.

brightproject, 2023-07-10:
I connected the I2C bus analyzer and began to see what was happening.
When I uploaded the [sketch](https://github.com/bolderflight/invensense-imu/blob/main/examples/arduino/mpu9250/i2c/i2c.ino)  to the `stm32f401ccu6` board, I connect it to port `com 6`.
Nothing happens until you hit reset on board.
Packet signals appear on the I2C bus.
[image]
The first three messages begin with writing the address of the sensor `0x69` to the `mpu9250` sensor and receiving two responses `0x6B` and `0x01`
[image]
These are the registers of the `mpu9250` sensor
```
PWR_MGMNT_1_ = 0x6B;
CLKSEL_PLL_ = 0x01;
```
Then 7 parcels follow.
[image]
```
Write [0x69] + ACK
0x6B + ACK
```
`PWR_MGMNT_1_ = 0x6B;`
```
Read [0x69] + ACK
0x01 + NAK
```
`INT_RAW_RDY_EN_ = 0x01;`
```
Write [0x69] + ACK
0x6A + ACK
0x20 + ACK
```
`USER_CTRL_ = 0x6A;`
`I2C_MST_EN_ = 0x20;`
This is also followed by parcels, as I understand it, setting up the `mpu9250` registers for work.
But everywhere the reading `Write [0x69]` from the sensor ends with `NAK`
The package ends with such parcels.
[image]
Well, in the port monitor there is a traditional inscription that the sensor has not been initialized.
[image]
Each line is a separate pressing of the RESET button on the BlackPill board and, accordingly, sending those packets that I described above.
Strange ... the library [SparkFun](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/examples/MPU9250_Basic/MPU9250_Basic.ino) is written generally for SAMD boards, even when compiling it, a warning appears that the library may not work ... but it works and works fine.
`WARNING: The SparkFun_MPU-9250-DMP_Arduino_Library-master library must run on SAMD architectures and may not be compatible with your stm32 board.`

The library [invensense-imu](https://github.com/bolderflight/invensense-imu/) does not issue any warnings, it compiles perfectly, communicates with the `mpu9250` sensor via `I2C` and does not work ... a paradox!

brightproject, 2023-07-11:
I continue experiments, since the author of the library does not want to figure it out.
I tried to connect the `MPU9260` module to the `ESP32` via `I2C`.
The situation is the same as with `stm32f401ccu6`.
As I already wrote, the `MPU9260` module works fine with the [SparkFun](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/examples/MPU9250_Basic/MPU9250_Basic.ino) library, and even gives out a quaternion - therefore, the [library](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/src/SparkFunMPU9250-DMP.h) in some way can "get through" to the magnetometer, because, without a magnetometer signal, the quaternion cannot be calculated ... I could be wrong.
After 3 days of torment with `I2C`, which for `MPU9260` is generally an "experimental" interface, it was decided to connect via `SPI`.
Connecting to `stm32` did this:
```
PA7 - SDA(MOSI)
PA6 - AD0(MISO)
PA5 - SCL(SCK)
PA4 - NCS(CS)
BY RESISTOR 10 kOm TO GND - FSYNC
```
SPI starter from command `0x6B`, `0x01` - `WriteRegister(PWR_MGMNT_1_, CLKSEL_PLL_)`...Further, there are also parcels on the SPI bus, but there is no normal operation.
[image]

```
bool Mpu9250::Begin() {
  imu_.Begin();
  /* 1 MHz for config */
  spi_clock_ = SPI_CFG_CLOCK_;
  /* Select clock source to gyro */
  if (!WriteRegister(PWR_MGMNT_1_, CLKSEL_PLL_)) {
    return false;
  }
  /* Enable I2C master mode */
  if (!WriteRegister(USER_CTRL_, I2C_MST_EN_)) {
    return false;
  }
  /* Set the I2C bus speed to 400 kHz */
  if (!WriteRegister(I2C_MST_CTRL_, I2C_MST_CLK_)) {
    return false;
  }
  /* Set AK8963 to power down */
  WriteAk8963Register(AK8963_CNTL1_, AK8963_PWR_DOWN_);
  /* Reset the MPU9250 */
  WriteRegister(PWR_MGMNT_1_, H_RESET_);
  /* Wait for MPU-9250 to come back up */
  delay(1);
  /* Reset the AK8963 */
  WriteAk8963Register(AK8963_CNTL2_, AK8963_RESET_);
  /* Select clock source to gyro */
  if (!WriteRegister(PWR_MGMNT_1_, CLKSEL_PLL_)) {
    return false;
  }
  /* Check the WHO AM I byte */
  if (!ReadRegisters(WHOAMI_, sizeof(who_am_i_), &who_am_i_)) {
    return false;
  }
  if ((who_am_i_ != WHOAMI_MPU9250_) && (who_am_i_ != WHOAMI_MPU9255_)) {
    return false;
  }
  /* Enable I2C master mode */
  if (!WriteRegister(USER_CTRL_, I2C_MST_EN_)) {
    return false;
  }
  /* Set the I2C bus speed to 400 kHz */
  if (!WriteRegister(I2C_MST_CTRL_, I2C_MST_CLK_)) {
    return false;
  }
  /* Check the AK8963 WHOAMI */
  if (!ReadAk8963Registers(AK8963_WHOAMI_, sizeof(who_am_i_), &who_am_i_)) {
    return false;
  }
  if (who_am_i_ != WHOAMI_AK8963_) {
    return false;
  }
  /* Get the magnetometer calibration */
  /* Set AK8963 to power down */
  if (!WriteAk8963Register(AK8963_CNTL1_, AK8963_PWR_DOWN_)) {
[... 66 lines cut ...]
  }
  /* Set the sample rate divider */
  if (!imu.ConfigSrd(19)) {
    Serial.println("Error configured SRD");
    //while(1) {}
  }
```
In order for the `MPU9250` module to start outputting data and get zeros again, except for the temperature sensor.
```
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    32.01
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    31.91
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    32.01
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    31.91
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    31.86
1       0       0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00    32.10
```
For the sake of interest, to make sure that the temperature data is fake, I cooled the MPU9250 chip with a freezer, and the temperature dropped to `minus 28 degrees Celsius` and then returned to room temperature.
[image]
Now the thought does not leave me - why does the `MPU9250` module transmit data on the temperature of its chip in the `I2C` bus and via `SPI`, but does not give out data on the accelerometer, gyroscope and magnetometer?
It's all about the [library](https://github.com/bolderflight/invensense-imu) and the @flybrianfly can't help.
Or just write that the library was created only for the [Tinsy](https://www.pjrc.com/teensy/) microcontroller and will not work anywhere else...
Although where does the microcontroller `Tinsy`?
The library does not work on ESP32 either.
It seems to me that the case is in incorrect registers and calls in `writeRegister` and `readRegisters` into [library](https://github.com/bolderflight/invensense-imu/blob/main/src/invensense_imu.cpp).
Can I somehow use this [library](https://github.com/bolderflight/invensense-imu) for the sensor `GY-LSM6DS3`?

flybrianfly, 2023-07-11:
Hi, rudeness will not be tolerated. I was traveling over the weekend to visit family and, always, my order of priority is to satisfy family and work obligations before working on open-source projects. Currently, we have a lot of work projects that are due soon - I've already triaged this issue and it's not something that can be easily fixed, since we don't even know what's wrong yet, so it will have to wait until I have a chance to dive deeper into the potential issue. This will likely be later this fall. I welcome a PR if you are able to solve the issue before then.

We use Teensy boards for development because they are well supported and follow the Arduino core guidelines; so I know that if I use a TwoWire method, it is working as intended. We've been working with these IMUs since they were released and the libraries have been working well for years now. Most issues with the library are due to:
1. Poor core implementations by board manufacturers. For example, ESP32 works, but required a modification to one of the TwoWire methods because the ESP32 core didn't implement *endTransmission(false)* correctly.
2. Trying to use the library with an unsupported IMU - this library only works with the MPU-6500 and MPU-9250. Others may be added in the future, but these are the InvenSense IMUs that are currently supported.
3. Counterfeit IMUs and clones.

In addition to the Teensy boards, this library has bee used on the [STM32L4 boards](https://github.com/GrumpyOldPizza/arduino-STM32L4), ESP32, and AVR. I'm sure there are others that I haven't personally tested, but in general it should work well across Arduino boards.

closed 2023-08-13 by brightproject

enoch-prince, 2024-05-08:
Still doesn't work on the ESP32 😒.
Arduino ESP32 Core: v2.0.13
ESP32 by Espressif Core: v2.0.11

flybrianfly, 2024-05-08:
@enoch-prince, can you give more information on the exact processor you're trying to use, the sensor, and the communication method? It should work if it's wired correctly and using a support chip that isn't counterfeit.